xoseperez/espurna #2221: Add support for LC Technology UART ESP-01 1/2/4CH relays

xates, 2020-04-11:
This family of boards ([example on AliExpress](https://www.aliexpress.com/item/32919575201.html)) is extremely common, they feature an ESP-01 that communicates with a STM8S103F MCU on the board through UART0 to control the relays. I think they were originally produced by LC Technology but there seem to exist clones from other vendors.

Espurna already supports - meaning there is a prebuilt image available for - a STM_RELAY (with `espurna-<version>-stm-relay.bin`), which is the version equipped with two relays, there are however versions with 1 and 4 relays as well.

It would be great to natively support these as well!
I see three potential solutions for this.

#### 1. Easiest and perhaps most user-friendly - Have an image for each version
Simply have a dedicated image for each of the 1, 2 and 4 channel versions.

Moreover, I would change the naming to make it easier to find the image (at first I thought this kind of board wasn't supported at all by Espurna, only searching through the issues before posting this one I discovered #1130 and then `espurna-<version>-stm-relay.bin`). I would include keywords in the name like `lctech` and `esp01` instead of `stm`  to make it more recognizable. The average user doesn't notice the presence of the STMicroelectronics chip - I didn't - it is more likely to find the image by the brand or the form factor of the ESP and the number of relays on the board.

So have something like:
- `espurna-<version>-lctech-esp01-1ch-relay.bin`
- `espurna-<version>-lctech-esp01-2ch-relay.bin`
- `espurna-<version>-lctech-esp01-4ch-relay.bin`

#### 2. Have a single image for all the 3 versions
Have a single image (like `espurna-<version>-lctech-esp01-stm-relay.bin`) and
- have an option on the web UI to choose how many relays the board has (1/2/4), or
- have initially four relays, and add the ability to remove or hide the excess ones.

The only advantage I see with this approach is having two image less to build, however I think it would also add some complexity as it requires to add an option to choose the amount of relays.

#### 3. Have single image for all UART relay boards
Have a single image with the ability to configure UART switches directly from a tab in the web UI in order to easily support all the different boards and circuits that might use UART - and not necessarily STMs or the same commands - to communicate with secondary MCUs that actually actuate the relays.

This would mean having an image (`espurna-<version>-uart-relay.bin`?) with a dedicated page with options to set:
- UART TX and RX pin
- baud rate
- stop bits
- possibly, also a configurable delay between commands (see #1519)

and to add relays, each with:
- command (data to send) to turn on
- command (data to send) to turn off

It would come with some presets - currently for LC Technology 1, 2 and 4 channel relays - that could be easily added once new devices are discovered.

This would make it extremely easy for users to use Espurna on similar devices - or older versions of this board which use a baud rate of 9600 instead of 115200 - without the need to build a custom image, and would make Espurna easily already compatible with potentially a lot of new different boards.

This is the most flexible solution, however it might be more complex to implement as it requires some reworking of the logic and the UI and will be less user-friendly **if** not equipped with the presets mentioned above.

**In conclusion**, I think the best course, at least at first, would be the first approach, as it can be implemented immediatly and quickly, but in my opinion this third idea should really be taken into consideration as it would be a very interesting feature to have on Espurna, maybe in a separate dedicated image in addition to the other three.

mcspr, 2020-04-12:
> 2. Have a single image for all the 3 versions
Have a single image (like espurna-<version>-lctech-esp01-stm-relay.bin) and

have an option on the web UI to choose how many relays the board has (1/2/4), or
have initially four relays, and add the ability to remove or hide the excess ones.
The only advantage I see with this approach is having two image less to build, however I think it would also add some complexity as it requires to add an option to choose the amount of relays.

Note the `DUMMY_RELAY_COUNT` flag, we do have a runtime setting that will override it on boot:
https://github.com/xoseperez/espurna/blob/2dcf1b2bd80ea2474275254bec722aa8905bf6c5/code/espurna/relay.ino#L1413-L1414
The way it is done is mostly historic reasons, we do need to prepend virtual relays before real ones and we have a pre-condition to ignore relays with GPIO_NONE for a pin thus, ignoring any RELAY#_PIN flag or anything we set via relayGPIO# settings. I wonder if there is some other way of specifying relays like every other setting, simply checking `moduleGPIO#` key value or compile time flag.

Delay option seems like another one to change based on user preference.
If this worked out with internal settings, it will work with WebUI when we add support for the specified key-values. Existing UI is tightly coupled with key-values, inputs elements reference them by name
*(most of the time. there are some small exceptions)*

mcspr, 2021-11-22:
Oops, this was missed in the relays update quite a while. Dummy relays are no longer used, and it is a different 'provider' for the specific relay.
https://github.com/xoseperez/espurna/blob/9d4ef1a38690252ff3e2ead31b14689ee9cb3d7f/code/espurna/relay.cpp#L898-L945
STM's relay ID is the one used by us. I wonder if it should be another setting for STM ID, allowing to mix different relays in-between

So, it's just a matter of using provider setting / build flag
https://github.com/xoseperez/espurna/blob/9d4ef1a38690252ff3e2ead31b14689ee9cb3d7f/code/espurna/config/hardware.h#L2146-L2147
https://github.com/xoseperez/espurna/blob/9d4ef1a38690252ff3e2ead31b14689ee9cb3d7f/code/espurna/relay.cpp#L462-L464

However, settings resort to enum->integer conversion, so it's `4` atm (`set relayProv0 4`, and etc.)
https://github.com/xoseperez/espurna/blob/9d4ef1a38690252ff3e2ead31b14689ee9cb3d7f/code/espurna/relay.cpp#L410
Might be a good idea to change to `stm` string instead. Update was stalled though, I intend to push 1.15 first and add those stringified settings later

Just defining the provider is enough, as it has no additional checks
https://github.com/xoseperez/espurna/blob/9d4ef1a38690252ff3e2ead31b14689ee9cb3d7f/code/espurna/relay.cpp#L2373-L2375