xoseperez/espurna #2223: Bug after upgrade to 1.14.2-dev

HilverinkJ, 2020-04-14:
[image]
[image]
[image]

- BME280 Pressure, not (correct) displayed anymore (IDX 233). 
- BME280 Humidity, nvalue is integer value, in domoticz log; MQTT invalid data recieved (IDX 235).
- BME280 UV, inserted the following code:
file domoticz.ino

   ` // svalue = UV;TEMP. Temperature (only for UV subtype TFA (3), else specify 0) Don't loose the `                                        `                                                             ";0" at the end - without it Domoticz database may corrupt.`
 `   } else if (MAGNITUDE_UVI == type) {`
`        String svalue = buffer;`
 `       svalue += ";0";`
 `       domoticzSend(key, 0, svalue.c_str());`

- Switch #0/1 nvalue is 0:1 svalue is empty (IDX 218/69)

[image]

mcspr, 2020-04-14:
Humidity values are mixed up :( svalue -> nvalue. domiticzSend should have values swapped, `nvalue` string first for the type and then `buffer` for the real sensor value.

Pressure thing is weird, does the `magnitudes` command output the correct value for reported & last?

mcspr, 2020-04-14:
Or not weird at all:
https://github.com/xoseperez/espurna/blob/2dcf1b2bd80ea2474275254bec722aa8905bf6c5/code/espurna/sensor.ino#L521-L524
This part causes pressure to be between 0 and 100, thus always displaying 100. Block can be safely removed

HilverinkJ, 2020-04-14:
OK. pressure issue solved now.

HilverinkJ, 2020-04-14:
> Humidity values are mixed up :( svalue -> nvalue. domiticzSend should have values swapped, nvalue string first for the type and then buffer for the real sensor value.

No not in this case. You have to convert nvalue to integer.

HilverinkJ, 2020-04-14:
> Humidity values are mixed up :( svalue -> nvalue. domiticzSend should have values swapped, nvalue string first for the type and then buffer for the real sensor value.

Not in this case.
This part is causing the trouble in both cases switch and humidity.

[image]

HilverinkJ, 2020-04-14:
[image]

mcspr, 2020-04-14:
What is the problem with empty svalue for the switch? I though domoticz does not care about it, ignoring any sent contents and only using nvalue.

And yeah, humidity is actually in the right place. I missed my own comment above, we do things differently than every other sensor output :/
```diff
@@ -227,7 +228,7 @@ void domoticzSendMagnitude(unsigned char type, unsigned char index, double value
             HUMIDITY_DRY
         );
         char svalue[2] = {status, '\0'};
-        domoticzSend(key, buffer, svalue);
+        domoticzSend(key, static_cast<int>(value), svalue);
     // Otherwise, send char string (nvalue is only for integers)
     } else {
         domoticzSend(key, 0, buffer);
```

HilverinkJ, 2020-04-14:
> What is the problem with empty svalue for the switch? I though domoticz does not care about it, ignoring any sent contents and only using nvalue.

Nothing, my mistake.

HilverinkJ, 2020-04-14:
You made me happy.

PR #2230 (2020-04-28): Various sensor fixes
fix #2223
fix #2219

- fix domoticz API floats in nvalue
- fix pressure constrain 0...100
- adjust BaseSensor api to allow different units per magnitude
- hlw8012 set energy in pre-read, getter returns last value
closed 2020-04-28 by mcspr