xoseperez/espurna #2232: More information on 'platformio_override.ini' usage.

davebuk, 2020-04-28:
I was looking through the files and noticed a reference to `platformio_override.ini`. I guess it allows an easier way to build custom setups without effecting the main repo files? I am was looking for more info on its use.

I already have a `custom.h` file but have been using the `ITEAD_SONOFF_BASIC` environment as a starting point and adjusting its definitions in the custom.h file. I also set `-DUSE_CUSTOM_H` to the default build flags and `platform = ${common.platform_latest}` so all builds use the latest version.

How does the `platform_override.ini` work with platformIO? Would I create the file similar to the details below and define in `custom.h` `#elif defined(MY_SONOFF_BUILD)`

platform_override.ini
```
[env:esp8266-1m-latest-base]
platform = ${common.platform_latest}
board = ${common.board_1m}
board_build.ldscript = ${common.ldscript_1m}

[env:my_sonoff_biuld]
extends = env:esp8266-1m-latest-base
src_build_flags = -DUSE_CUSTOM_H
```

mcspr, 2020-04-29:
- `[env:esp8266-1m-latest-base]` is redundant, 'extends =' will read the base file (see platformio_ota.ini that handles old -ota envs)
-  `env:` name isn't used to provide hardware flag, you need to set it yourself
`src_build_flags = -DMY_SONOFF_BUILD -DUSE_CUSTOM_H`
- also note that values will not be merged, you are always overriding the existing value. for example, to re-use `src_build_flags` from other env, you would need to have something like `src_build_flags = ${env:itead-sonoff-basic.src_build_flags} -DSOME_OTHER_FLAG`

davebuk, 2020-04-29:
So for my custom Sonoff Basics I could have:

`platform_override.ini`
```
[env:my_sonoff_build01]    // My custom Sonoff config
extends = env:esp8266-1m-latest-base
src_build_flags = ${env:itead-sonoff-basic.src_build_flags} -DUSE_CUSTOM_H

[env:my_sonoff_build02]    // Another custom Sonoff config
extends = env:esp8266-1m-latest-base
src_build_flags = ${env:itead-sonoff-basic.src_build_flags} -DUSE_CUSTOM_H

[env:my_sonoff_build03]    // Yet another custom Sonoff config
extends = env:esp8266-1m-latest-base
src_build_flags = ${env:itead-sonoff-basic.src_build_flags} -DUSE_CUSTOM_H
```

Then in `custom.h` I would have:
```
#elif defined(MY_SONOFF_BUILD01)
#define
#define
#etc......

#elif defined(MY_SONOFF_BUILD02)
#define
#define
#etc......

#elif defined(MY_SONOFF_BUILD03)
#define
#define
#etc......
```
Would the environments:  `my_sonoff_build01`, `my_sonoff_build02` and `my_sonoff_build03` then appear as build options in VSCode/PlatformIO build list and I can click on the one I want to build?

mcspr, 2020-04-30:
Solved? We can add a more concise comment to the platformio.ini regarding the override.ini rules, if you have some ideas about that

davebuk, 2020-04-30:
Yep, all seems to work. I only added `MY_BUILD_01` to the two files, but it built a firmware for the Sonoff with the options laid out in my `custom.h` file.

Would an entry in the Wiki would be better? It would allow for more examples without adding extra lines to `platformio.ini`.

mcspr, 2020-04-30:
Examples are better in wiki, true. Also see:
- https://docs.platformio.org/en/latest/projectconf/section_platformio.html#extra-configs
- `pio project config` text output (or with --json-output & filter with something like jq)

.ini might need some additional lines:
- override.ini `[common] key = value` takes precedence over main .ini key-value
- we cannot refer to 'previous' version of the key, there is only one value across the config
- key=values are replacing things, not merging them (i.e. what I referred to in the build_flags example above)

davebuk, 2020-05-01:
Are you referring to the lines:
```
[env:XXX_XXX]
extends =
src_build_flags =
```
In an example, are these two lines and the [env:] details basically the minimum required in `platformio_override.ini` file?

mcspr, 2020-05-03:
Yes. extends puts board, ldscript and platform / platform_packages, which correctly set Core version and flash size used by the build. src flags are only supposed to configure our side of things (e.g. add / ..._SUPPORT or set some pins, or helping out with custom.h)
We implicitly overlay [env] btw, things like build_flags and libraries are coming from there.

davebuk, 2020-05-03:
For new people looking to build espurna using PlatformIO, is it worth having an example file in the repository rather than examples inside `platformio.ini`? I having been trying out https://github.com/1technophile/OpenMQTTGateway to act as a 433MHz to MQTT gateway. They use a similar override file to allow individual builds.

A `custom.h.example` may also be useful?

mcspr, 2020-05-04:
Multiple files is a tricky subject, since we overlay everything into everything. So, in case you have two env:my-env entries in two _env.ini, which one is used exactly?

Please check out changes at the bottom https://github.com/xoseperez/espurna/wiki/PlatformIO

davebuk, 2020-05-04:
It was more the idea of having a `platformio_override.ini.example` and `custom.h.example` as a starting point for others. They would just copy, paste and rename, dropping the `.example` off their working copy.

Wiki looks good. It'll give myself a reminder of what to do in the future.

closed 2020-05-04 by davebuk

mcspr, 2020-05-05:
If you have a good examples in mind, can you please do a PR?
(otherwise, I have some very small examples in mind to describe use-case we already discussed above)

davebuk, 2020-05-05:
I'll give it a go but you'll need to check I've not missed anything out. It might be a week or so tough.

mcspr, 2020-05-05:
Thanks! No hurry.