xoseperez/espurna #2237: Add support for NEO Coolcam 16A Power Monitoring Plug

spike411, 2020-05-01:
Hello!

Could you please add support for the 16A power monitoring version(s) of the Neo COOLCAM plug?

https://www.aliexpress.com/item/32963796295.html

There are actually two revisions:
https://templates.blakadder.com/neocoolcam_16a.html
https://templates.blakadder.com/neocoolcam_16a_v2.html

Please note: the power metering function requires calibration which can be a bit impractical for regular users who don't have a multimeter at home (then again, they can order one from Aliexpress along with their Coolcam plugs :)).

I have the older revision to test with, and I will be ordering new ones (hopefully the new revision arrives).

Best regards
Lukáš

mcspr, 2020-05-01:
Hey,

Checkout hardware.h, I do see that we match both pins with HYKKER_SMART_HOME_POWER_PLUG
https://github.com/xoseperez/espurna/blob/9d2534f6414241e14fe40285c748e85a86cb8a4b/code/espurna/config/hardware.h#L4466
We don't have any distinction between HLW8012 and BL chips in the code, but templates do specify different CF modes. May not be a problem, though.

WhoCares1337, 2020-05-26:
My 16A Power Monitoring NEO Coolcam seems to work fine with espurna-1.14.1-hykker-smart-home-power-plug.bin. Thank you very much!

closed 2020-08-01 by stale[bot]