xoseperez/espurna #2244: RPN Rules: relay0 variable not populated after boot.

pezinek, 2020-05-05:
Hi,

Seems that some "basic" variables for the RPN rules like $relay0 are not populated after boot.

how to reproduce:

1) compile espurna 1.14.1 similarly to this:

```
ESPURNA_FLAGS="$ESPURNA_FLAGS -DDHT_SUPPORT=1 -DDHT_PIN=14 -DBUTTON2_MODE=7 -DBUTTON2_PIN=4 -DBUTTON2_RELAY=2 -DRELAY2_PIN=2 -DRPN_RULES_SUPPORT=1" \
pio run -t upload -e wemos-d1mini-relayshield --upload-port 192.168.1.250
```

2) boot the device, don't touch any buttons/relays and enter into the console something like:

```
RPN.VARS
[714197] [RPN] Variables:
[714197]    temperature0: 24.90
[714197]    humidity0: 38.00
[714198]    relay1: 0.00
[714198] +OK
```
note that relay0 is missing

3) switch relay0 on/off and repeat the above:

```
RPN.VARS
[731269] [RPN] Variables:
[731270]    temperature0: 24.90
[731270]    humidity0: 38.00
[731270]    relay1: 0.00
[731270]    relay0: 0.00
[731271] +OK
```
note that the relay0 is present now

.... the above is causing that rules using relay0 are not working after boot until after relay0 is toggled manually at least once.

(just for the reference the RPN rule I'm talking about is this one:
`$temperature0 30 35 cmp3 1 + $relay1 $relay1 $relay0 or 1 3 index 0 relay`)

mcspr, 2020-05-07:
Looks like a 2-parter, I'll just explain the code paths we use:
When setup() is called, we call relaySetup() at some point before rpnSetup(). Relay setup code also calls relayLoop() and thus changes relay status before rpn context is even initialized. We can try to remove loop call from setup and wait until we exit setup() and enter loop(), thus avoiding ordering issue.

pezinek, 2020-05-09:
Actually I tried to move the rpnSetup() before relaySetup() and it seems to be working just fine.
Would that be acceptable fix ? (I'm not really familiar with the espurna code so I can't really foresee all the eventual consequences of this approach)

```diff --git a/code/espurna/espurna.ino b/code/espurna/espurna.ino
index 03fe355..3d6870b 100644
--- a/code/espurna/espurna.ino
+++ b/code/espurna/espurna.ino
@@ -159,6 +159,10 @@ void setup() {
     #if LIGHT_PROVIDER != LIGHT_PROVIDER_NONE
         lightSetup();
     #endif
+    // rpnSetup must be called before relaySetup to initialize $relayN variables
+    #if RPN_RULES_SUPPORT
+        rpnSetup();
+    #endif
     relaySetup();
     #if BUTTON_SUPPORT
         buttonSetup();
@@ -227,9 +231,6 @@ void setup() {
     #if SCHEDULER_SUPPORT
         schSetup();
     #endif
-    #if RPN_RULES_SUPPORT
-        rpnSetup();
-    #endif
     #if UART_MQTT_SUPPORT
         uartmqttSetup();
     #endif
```

mcspr, 2020-05-11:
Yes, I missed that option, thanks for noticing. Probably makes more sense right now and it should not break anything (unlike relay stuff that is slightly more that 3 lines).

closed 2020-05-12 by mcspr