xoseperez/espurna #2256: Intermittent web interface bug

nuxeh, 2020-05-19:
**Bug description**
Sometimes when connecting to SoftAP, the web interface will not be available, with both Firefox and mobile Chrome trying to download an octet stream.

**Steps to reproduce**
Flash an ESPurna image to a device, and connect to SoftAP.

**Expected behavior**
Web interface available.

**Device information**
```
[402024] [MAIN] ESPURNA 1.14.2-dev (d5dc1e02)
[402024] [MAIN] xose.perez@gmail.com
[402024] [MAIN] http://tinkerman.cat

[402024] [MAIN] CPU chip ID: 0x251079
[402025] [MAIN] CPU frequency: 80 MHz
[402025] [MAIN] SDK version: 1.5.3(aec24ac9)
[402025] [MAIN] Core version: 2.3.0
[402026] [MAIN] Core revision: 9826c6d
[402026] [MAIN] Build time: 1589451558
[402026] 
[402027] [MAIN] Flash chip ID: 0x144051
[402030] [MAIN] Flash speed: 40000000 Hz
[402030] [MAIN] Flash mode: DOUT
[402030] 
[402031] [MAIN] Flash size (CHIP)   :  1048576 bytes /  256 sectors (   0 to  255)
[402034] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[402034] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[402034] [MAIN] Firmware size       :   511552 bytes /  125 sectors (   1 to  125)
[402034] [MAIN] Max OTA size        :   512000 bytes /  125 sectors ( 126 to  250)
[402035] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[402035] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[402036] 
[402036] [MAIN] EEPROM sectors: 251, 250
[402037] [MAIN] EEPROM current: 250
[402037] 
[402038] [MAIN] EEPROM:  4096 bytes initially |    74 bytes used ( 1%) |  4022 bytes free (98%)
[402038] [MAIN] Heap  : 31736 bytes initially | 21848 bytes used (68%) |  9888 bytes free (31%)
[402039] [MAIN] Stack :  4096 bytes initially |  1424 bytes used (34%) |  2672 bytes free (65%)
[402040] 
[402040] [MAIN] Boot version: 31
[402040] [MAIN] Boot mode: 1
[402040] [MAIN] Last reset reason: Power on
[402041] [MAIN] Last reset info: flag: 0
[402041] 
[402042] [MAIN] Board: BLITZWOLF_BWSHPX_V23
[402043] [MAIN] Support: API BROKER BUTTON DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT LED MDNS_SERVER MQTT NTP SCHEDULER SENSOR TELNET TERMINAL WEB 
[402043] [MAIN] OTA: ARDUINO ASYNCTCP WEB 
[402043] [MAIN] Sensors: HLW8012 
[402044] [MAIN] WebUI image: SENSOR
[402044] 
[402279] [MAIN] Firmware MD5: 194654f75903867cb20d60b1ee5de0f9
[402281] [MAIN] Power: 3185 mV
[402281] [MAIN] Power saving delay value: 10 ms
[402281] 
```

**Tools used**
* NixOS unstable (Linux)
* Firefox 76.0.1 (64-bit)
* PlatformIO, version 4.1.0

**Additional context**
This has presented itself anecdotally when flashing via tuya convert.

mcspr, 2020-05-20:
ESPAsyncWebServer may have some issues with chunked encoding stream, which *may* result in Firefox treating it as something else? Does it download the index as a unknown blob or something else?

Can't really tell from just the description, seeing raw traffic from tcpdump / wireshark will likely show the issue Firefox encounters. fwiw, low memory condition should just show a blank page, which will return to normal when browser instance is closed (as it sometimes likes to keep lingering connections). Overall bad wireless connection is another possible reason (but idk what exactly we should do about it), not sure what else to blame.

nuxeh, 2020-05-20:
I didn't inspect the file it attempted to download, as i recall it was empty.

I realise this is a tricky thing to debug, I do recall reading about others having the same issue, which i couldn't find again, so I thought i'd raise the issue to see if there's any interest.

It happened in multiple browsers, a fresh flash, in SoftAP mode, no distance between the computer and the device, and first connection to the device, so i don't think the scenarios you put forward are the cause.

It's not a big issue since it seems to snap out of it after power cycling, and leaving it off for 10 seconds in this case, but I think it's certainly a bug, albeit a minor one, which is a slight harm to user experience. It's not the first time I've seen it happen either, flashing ESPurna across 10-20 devices.

Next time I'll get a tcpdump, if it happens again.

mcspr, 2020-05-21:
Trying to search for 'octet steam' keywords:
https://github.com/xoseperez/espurna/issues/24#issuecomment-360133283
We do indeed have a bug with networking when using custom builds  with the 2.3.0 Core, see:
https://github.com/xoseperez/espurna/issues/1610
`pio run -t build-lwip` / using `latest-base` or `git-base` as environment (or changing extends=...) will help.

---

Do you happen to save the original firmware from before the tuya-convert?
I wonder if it is not specifically a sw bug, but something about the state we leave ourselves after tuya-convert is done, requiring power cycle.

nuxeh, 2020-05-24:
Interesting.

It happened again, and I got a couple of packet captures:

- Response as octet-stream
- After hard power cycle, nothing on first attempt, but web UI on refreshing the page

[espurna-pcap.zip](https://github.com/xoseperez/espurna/files/4673344/espurna-pcap.zip)

I can get the firmware backup easily enough if it would be of use.

mcspr, 2020-05-25:
Not really sure why it tries to fragment stuff into a 536-sized packets. I had tried Windows PC connecting to a softAP, but I only see 1460 packets sent. 
I'd expect latest Core builds work? If this is somehow related to the #1610, it does fix a lot of stuff (...afaik, fragmentation is kind of broken with old Core / lwip it includes.) 

I also wonder if it is somehow related to Linux? From not-so-recent memory, https://github.com/xoseperez/espurna/issues/1595 mentioned similar issues. I'll check it out what Ubuntu live session does.
Do you have something changed from 'net.ipv4' defaults via `sysctl`?

nuxeh, 2020-07-25:
@mcspr I'd completely forgotten to follow up. I haven't changed any `sysctl` settings, but this is a dump of the default values on the system I was using (running NixOS) for reference:

```
/proc/sys/net/ipv4/cipso_cache_bucket_size 10
/proc/sys/net/ipv4/cipso_cache_enable 1
/proc/sys/net/ipv4/cipso_rbm_optfmt 0
/proc/sys/net/ipv4/cipso_rbm_strictvalid 1
/proc/sys/net/ipv4/fib_multipath_hash_policy 0
/proc/sys/net/ipv4/fib_multipath_use_neigh 0
/proc/sys/net/ipv4/fib_sync_mem 524288
/proc/sys/net/ipv4/fwmark_reflect 0
/proc/sys/net/ipv4/icmp_echo_ignore_all 0
/proc/sys/net/ipv4/icmp_echo_ignore_broadcasts 1
/proc/sys/net/ipv4/icmp_errors_use_inbound_ifaddr 0
/proc/sys/net/ipv4/icmp_ignore_bogus_error_responses 1
/proc/sys/net/ipv4/icmp_msgs_burst 50
/proc/sys/net/ipv4/icmp_msgs_per_sec 1000
/proc/sys/net/ipv4/icmp_ratelimit 1000
/proc/sys/net/ipv4/icmp_ratemask 6168
/proc/sys/net/ipv4/igmp_link_local_mcast_reports 1
/proc/sys/net/ipv4/igmp_max_memberships 20
/proc/sys/net/ipv4/igmp_max_msf 10
/proc/sys/net/ipv4/igmp_qrv 2
/proc/sys/net/ipv4/inet_peer_maxttl 600
/proc/sys/net/ipv4/inet_peer_minttl 120
/proc/sys/net/ipv4/inet_peer_threshold 65664
/proc/sys/net/ipv4/ip_default_ttl 64
/proc/sys/net/ipv4/ip_dynaddr 0
/proc/sys/net/ipv4/ip_early_demux 1
/proc/sys/net/ipv4/ip_forward 0
/proc/sys/net/ipv4/ip_forward_update_priority 1
/proc/sys/net/ipv4/ip_forward_use_pmtu 0
/proc/sys/net/ipv4/ipfrag_high_thresh 4194304
/proc/sys/net/ipv4/ipfrag_low_thresh 3145728
/proc/sys/net/ipv4/ipfrag_max_dist 64
/proc/sys/net/ipv4/ipfrag_secret_interval 0
/proc/sys/net/ipv4/ipfrag_time 30
/proc/sys/net/ipv4/ip_local_port_range 32768 60999
/proc/sys/net/ipv4/ip_local_reserved_ports 
/proc/sys/net/ipv4/ip_nonlocal_bind 0
/proc/sys/net/ipv4/ip_no_pmtu_disc 0
/proc/sys/net/ipv4/ip_unprivileged_port_start 1024
/proc/sys/net/ipv4/ping_group_range 0 2147483647
/proc/sys/net/ipv4/raw_l3mdev_accept 1
/proc/sys/net/ipv4/tcp_abort_on_overflow 0
/proc/sys/net/ipv4/tcp_adv_win_scale 1
/proc/sys/net/ipv4/tcp_allowed_congestion_control reno
/proc/sys/net/ipv4/tcp_app_win 31
/proc/sys/net/ipv4/tcp_autocorking 1
/proc/sys/net/ipv4/tcp_available_congestion_control reno
/proc/sys/net/ipv4/tcp_available_ulp 
/proc/sys/net/ipv4/tcp_base_mss 1024
/proc/sys/net/ipv4/tcp_challenge_ack_limit 1000
/proc/sys/net/ipv4/tcp_comp_sack_delay_ns 1000000
/proc/sys/net/ipv4/tcp_comp_sack_nr 44
/proc/sys/net/ipv4/tcp_congestion_control reno
/proc/sys/net/ipv4/tcp_dsack 1
/proc/sys/net/ipv4/tcp_early_demux 1
/proc/sys/net/ipv4/tcp_early_retrans 3
/proc/sys/net/ipv4/tcp_ecn 2
/proc/sys/net/ipv4/tcp_ecn_fallback 1
/proc/sys/net/ipv4/tcp_fack 0
/proc/sys/net/ipv4/tcp_fastopen 1
/proc/sys/net/ipv4/tcp_fastopen_blackhole_timeout_sec 3600
/proc/sys/net/ipv4/tcp_fastopen_key 
/proc/sys/net/ipv4/tcp_fin_timeout 60
/proc/sys/net/ipv4/tcp_frto 2
/proc/sys/net/ipv4/tcp_fwmark_accept 0
/proc/sys/net/ipv4/tcp_invalid_ratelimit 500
/proc/sys/net/ipv4/tcp_keepalive_intvl 75
/proc/sys/net/ipv4/tcp_keepalive_probes 9
/proc/sys/net/ipv4/tcp_keepalive_time 7200
/proc/sys/net/ipv4/tcp_l3mdev_accept 0
/proc/sys/net/ipv4/tcp_limit_output_bytes 1048576
/proc/sys/net/ipv4/tcp_low_latency 0
/proc/sys/net/ipv4/tcp_max_orphans 65536
/proc/sys/net/ipv4/tcp_max_reordering 300
/proc/sys/net/ipv4/tcp_max_syn_backlog 1024
/proc/sys/net/ipv4/tcp_max_tw_buckets 65536
/proc/sys/net/ipv4/tcp_mem 190209 253615 380418
/proc/sys/net/ipv4/tcp_min_rtt_wlen 300
/proc/sys/net/ipv4/tcp_min_snd_mss 48
/proc/sys/net/ipv4/tcp_min_tso_segs 2
/proc/sys/net/ipv4/tcp_moderate_rcvbuf 1
/proc/sys/net/ipv4/tcp_mtu_probe_floor 48
/proc/sys/net/ipv4/tcp_mtu_probing 0
/proc/sys/net/ipv4/tcp_no_metrics_save 0
/proc/sys/net/ipv4/tcp_notsent_lowat 4294967295
/proc/sys/net/ipv4/tcp_orphan_retries 0
/proc/sys/net/ipv4/tcp_pacing_ca_ratio 120
/proc/sys/net/ipv4/tcp_pacing_ss_ratio 200
/proc/sys/net/ipv4/tcp_probe_interval 600
/proc/sys/net/ipv4/tcp_probe_threshold 8
/proc/sys/net/ipv4/tcp_recovery 1
/proc/sys/net/ipv4/tcp_reordering 3
/proc/sys/net/ipv4/tcp_retrans_collapse 1
/proc/sys/net/ipv4/tcp_retries1 3
/proc/sys/net/ipv4/tcp_retries2 15
/proc/sys/net/ipv4/tcp_rfc1337 0
/proc/sys/net/ipv4/tcp_rmem 4096 131072 6291456
/proc/sys/net/ipv4/tcp_rx_skb_cache 0
/proc/sys/net/ipv4/tcp_sack 1
/proc/sys/net/ipv4/tcp_slow_start_after_idle 1
/proc/sys/net/ipv4/tcp_stdurg 0
/proc/sys/net/ipv4/tcp_synack_retries 5
/proc/sys/net/ipv4/tcp_syncookies 1
/proc/sys/net/ipv4/tcp_syn_retries 6
/proc/sys/net/ipv4/tcp_thin_linear_timeouts 0
/proc/sys/net/ipv4/tcp_timestamps 1
/proc/sys/net/ipv4/tcp_tso_win_divisor 3
/proc/sys/net/ipv4/tcp_tw_reuse 2
/proc/sys/net/ipv4/tcp_tx_skb_cache 0
/proc/sys/net/ipv4/tcp_window_scaling 1
/proc/sys/net/ipv4/tcp_wmem 4096 16384 4194304
/proc/sys/net/ipv4/tcp_workaround_signed_windows 0
/proc/sys/net/ipv4/udp_early_demux 1
/proc/sys/net/ipv4/udp_l3mdev_accept 0
/proc/sys/net/ipv4/udp_mem 380421 507230 760842
/proc/sys/net/ipv4/udp_rmem_min 4096
/proc/sys/net/ipv4/udp_wmem_min 4096
/proc/sys/net/ipv4/xfrm4_gc_thresh 32768
```

mcspr, 2020-07-25:
Re: `[402025] [MAIN] Core version: 2.3.0`
Have you tried building the fw and using -latest as base environment?

mcspr, 2020-09-27:
> Re: [402025] [MAIN] Core version: 2.3.0
Have you tried building the fw and using -latest as base environment?

I believe the issue should be fixed in the latest development versions. Closing for now, but please check it out!

closed 2020-09-27 by mcspr