xoseperez/espurna #2258: "less than" or "greater than" filter for build numbers in nofuss server. (versions.json)

sfilinto, 2020-05-24:
**Is your feature request related to a problem? Please describe.**
i regularly build espurna firmware enabling & disabling different options as needed & distribute these to my devices OTA using nofuss. In a scenario of updating to a newer build of an existing version, it enters into an update loop.

if i knew the build number in advance (which is the unix time the binary was created at) i could use the "build_not" filter to stop once the build number is reached. however this is not an options as there being no easy way to extract the "build" number from the .bin file.

(without knowing the exact build number the "build_not" filter never matches the just updated firmware thus on reboot, the espurna will again update itself with the same firmware, thus going into a loop.)

Note: Version numbers can be filtered by "less than" or "greater than" filters, not build numbers.

**Describe the solution you'd like**
a versions.json filter which allows filtering by "less than" OR "greater than" a specified build number.

**Describe alternatives you've considered**
a method / standalone script of extracting the "build" number from the .bin file prior to updating the firmware would allow the build_not filter to be used.

mcspr, 2020-05-26:
Have you checked out what server code https://bitbucket.org/xoseperez/nofuss/src/master/server/php/ does? I'd recommend not using bitbucket webui and just clone it locally, the gist of the code relies on the PHP internal func https://www.php.net/manual/en/function.version-compare.php
Is this something we need to change in the server?

> a method / standalone script of extracting the "build" number from the .bin file prior to updating the firmware would allow the build_not filter to be used.

- Is it ok to rely on build time when re-building the code? We would get the exact same code with updated timestamp, but device will upgrade anyway.

- Can something external be used instead? e.g., see https://github.com/mcspr/espurna-nightly-builder/releases; I only use git tag as distinction, but if using something like `git-describe --tags` which also includes commit number relative to the latest tag, each new version will greater-than the previous one.

m-kozlowski, 2020-05-31:
This is what I'm using
```
diff --git a/server/php/src/routes.php b/server/php/src/routes.php
index f302c96..e40baea 100644
--- a/server/php/src/routes.php
+++ b/server/php/src/routes.php
@@ -53,8 +53,12 @@ $app->get('/', function($request, $response, $args) {
             if (!compare($entry, 'le', '>=', $version)) continue;
             if (!compare($entry, 'eq', '==', $version)) continue;

+            if (!compare($entry, 'build_lt', '>', $build)) continue;
+            if (!compare($entry, 'build_gt', '<', $build)) continue;
+
             // If build_not is defined in the JSON file, then the request must also have defined the build and they must NOT match
             $_build = get($entry['origin']['build_not'], "*");
+
             if ($_build != "*") {
                 if ($build == "") continue;
                 if ($build == $_build) continue;

```
```
    {
        "origin": {
            "mac": "*",
            "device": "ESPURNA_BWSHPX",
            "gt": "*",
            "le": "1.14.0",
            "build_lt": "1575188054"
        },
        "target": {
            "version": "1.14.0",
            "firmware": "/firmware/espurna-1.14.0-blitzwolf-bwshpx.bin",
            "spiffs": ""
        }
    },

```

m-kozlowski, 2020-05-31:
the tricky part is knowing build number before flash.
Well, I'm not proud of this solution, but that's what I do:
```
#!/bin/bash

for fw in $* ; do

    ts=$(strings $fw | grep -Pzo "\w{3}\s{1,2}\d{1,2}\s(2019|2020)\n\d{2}:\d{2}:\d{2}" | tr '\n' ' ' | tr -d '\0')
if [ -z "$ts" ] ; then
    echo $fw: ERROR
    exit 1
else
    ts="$ts UTC"
fi

echo -e $(basename $fw)"\t"$(date +%s -d "$ts")

done | column -t
```
```
$ ./get_build_ts.sh ../firmware/espurna-1.14.2-dev/espurna-1.14.2-dev-blitzwolf-bwshpx.bin
espurna-1.14.2-dev-blitzwolf-bwshpx.bin  1588549376

```

sfilinto, 2020-05-31:
> This is what I'm using
>
> ```
> diff --git a/server/php/src/routes.php b/server/php/src/routes.php
> index f302c96..e40baea 100644
> --- a/server/php/src/routes.php
> +++ b/server/php/src/routes.php
> @@ -53,8 [...]

Thanks. This sounds simple, will give it a try.

sfilinto, 2020-05-31:
> the tricky part is knowing build number before flash.
> Well, I'm not proud of this solution, but that's what I do:
>
> ```
> #!/bin/bash
>
> for fw in $* ; do
>
>     ts=$(strings $fw | grep -Pzo [...]

i think i understand the risks. this works for now. i build my binaries not more often then once a week, so i can safely estimate a range and use > or < in the build number.

mcspr, 2020-05-31:
I believe that reads part of the webserver's last-modified header?

I do remember suggesting in gitter changing __UNIX_TIMESTAMP__ to a symbol instead of a macro. Either:
- add `extern "C"` so it does not get mangled
- place it in a unique ELF section via `__attribute__((section("name")))`
Then it is just a matter of reading ELF format, seeking symbol and reading raw file location of where the those 4 bytes of unsigned long number are.
(either `objdump -r -j name`, `pyelftools` scripting which comes with PlatformIO, or something else)

sfilinto, 2020-06-08:
> I believe that reads part of the webserver's last-modified header?
> 
> I do remember suggesting in gitter changing __UNIX_TIMESTAMP__ to a symbol instead of a macro. Either:

Yes, you did suggest this on gitter. Unfortunately i dont have the skillet to figure out your instructions.
if the steps for filtering build number provided by @m-kozlowski  work, knowing the build number should be unneeded.

sfilinto, 2020-09-08:
@m-kozlowski happy to report this works as expected. the changes i did were:

in /server/php/src/routes.php
added 2 lines
```
if (!compare($entry, 'build_lt', '>', $build)) continue;
if (!compare($entry, 'build_gt', '<', $build)) continue;
```

in /var/www/html/nofuss/server/php/data/versions.json
replaced
```
build_not
```
with
```
build_lt
```
the value was a unix time just before i built the binaries.

this way i can now update the build without know the build number in advance & not go into an update loop.

closed 2020-09-08 by sfilinto