xoseperez/espurna #2259: Resources for learning to code for espurna etc.

davebuk, 2020-05-24:
I'd like to learn more about how the code is written to make firmware like espurna run. Are there some tutorials or other resources anyone could recommend to give myself a good starting point in how the code is built up?

Thanks.

mcspr, 2020-05-25:
In general, we are dealing with 3 layers from the runtime side:
- Arduino stuff, that is supposed to be cross-platform way to write MCU software:
https://www.arduino.cc/reference/en
- ESP8266 Core implements Arduino + it's specific extensions. C++ was chosen as the base for Arduino, so ESP8266 Core is written in that for the most part. In case of esp8266, right now it is C++11:
https://github.com/esp8266/Arduino
- We have a pretty minimal API on top of that:
https://github.com/xoseperez/espurna/wiki/3rd-Party-Plugins
https://en.cppreference.com/

tbh, I don't even know where to start with c++ part. I'd think some basic understanding of C is a must, then it is a matter of searching google -> cppreference | stackoverflow :)

davebuk, 2020-05-25:
Thanks for the info. I'd started reading 'Arduino for dummies'. I just need some time to finish it. I guess the ESP 8266 core stuff builds on that.

Sounds like the C++ will have to wait for another time! Thanks for the pointers.

closed 2020-05-25 by davebuk