xoseperez/espurna #2260: HTTP/REST/RPC execute Terminal command?

mindtripper, 2020-05-26:
Hi!

Can HTTP/REST/RPC be used to execute any Terminal command?

I ask because I would like to set mqttServer to a different IP/hostname using HTTP (REST/RPC).
In addition the following commands are also of interest to use the same way - mqtt.reset and ha.send.

If this is possible. Could you please provide an example?

Thank you in advance.

mcspr, 2020-05-26:
After https://github.com/xoseperez/espurna/commit/b8fc8cd1fd9359705bcc84ac53cdfae817726170, by:
- Using MQTT; adding the `TERMINAL_MQTT_SUPPORT` build flag set to 1 and sending something to the `<$mqttTopic>/cmd/set`. Right now it is only one command at a time, e.g. `set a foo\r\nset b bar\r\n` will only execute `set a foo`
- Using HTTP; add `TERMINAL_WEB_AP_SUPPORT` build flag set to 1 and enable API (`apiEnable 1`, `apiKey` should not be empty). Works like API, endpoint is `/api/cmd?apikey=...&line=...`. Same rule for line breaks, only the first command will be called. e.g.
```
$ curl -XPUT -v 'http://192.168.4.1/api/cmd' --data 'apikey=12345678&line=commands'
```
Should show available terminal commands

mindtripper, 2020-05-26:
This is exactly what I was asking for. Very much appreciated!

Next step is to build my own firmware with the suggested build flags. Or do pre-existing firmware builds with this enabled exist?

/Christer

mcspr, 2020-05-26:
Yep, check out the wiki:
https://github.com/xoseperez/espurna/wiki/PlatformIO

btw I have also forgot about the configuration endpoint, specifically made for dealing with settings:
https://github.com/xoseperez/espurna/wiki/Configuration#direct-http-settings-api
Right now it will overwrite the whole configuration. I wonder if we can add `PATCH` handler for partial update, only tricky part is deleting keys when using that logic.

mindtripper, 2020-05-27:
I've read it on the wiki before but forgot all about it :/ Thank you for pointing it out.
This solves my use-case, but I'm still planning on building my own firmware with TERMINAL_WEB_AP_SUPPORT to see how it works.

Thank you for taking the time to help me out!

mindtripper, 2020-06-05:
Closing. As the question has been answered.

closed 2020-06-05 by mindtripper