xoseperez/espurna #2264: POW - minimum power value

CrispinP, 2020-05-27:
**Is your feature request related to a problem? Please describe.**
New feature request 

**Describe the solution you'd like**
I'm sonoff pow, can we have a min-value setting for power. I use one to measure solar generation (grid-tie) and at night it fluctuates around 1-2W which I assume is noise. 
Can we have a box which is report-zero-if-equal-to-or-below. 

**Describe alternatives you've considered**
I do the same in openhab, use proxy item and check in rule but it's "overkill" 

Thanks.

mcspr, 2020-06-05:
Do you mean a setting to disable value 'report' when it is below a certain threshold?
e.g. for `MAGNITUDE_POWER_ACTIVE`, we can have a setting like  `pwrPMinThresh => 50` and it will only report when value is more that 50? Or still report, but set to 0?

CrispinP, 2020-06-05:
I would want it to report zero so maybe have the report_value as a variable?

To have it stop reporting it would not remove the noise.

mcspr, 2020-06-07:
As a variable?

Regarding values, we can put a different check like here, when relays are off it sets value to 0
https://github.com/xoseperez/espurna/blob/ba9b4a4503af014cb3652ff3c99897682f42daf7/code/espurna/sensor.cpp#L2610
It is slightly more complicated, because we do need specific value to limit it by?

Name of this feature / flag / setting is another problem :)

mcspr, 2020-06-08:
Slightly dumb question, but we do only care about active power? Not any other measurements related to power monitor? i.e. we can check every measurement in 'power' category - reactive, apparent, factor, voltage, energy_delta, and have a specific configuration setting for each one.

At least now I got an idea how this fits into sensor read-report loop.

mcspr, 2020-07-03:
Please check out #2290
As described in the commit message, there is a new setting that will check sensor value against a certain threshold (non inclusive). With only one device, index is 0.

Altough - there is no UI config for this, yet. And no build setting, only runtime configuration.

PR #2290 (2020-07-03): sns: more magnitude settings
- introduce zero threshold setting to reset value to 0 below a certain point (resolve #2264)
for example, `set pwrPZeroThreshold0 10.0` for active power (or `pwrQ...` for apparent, `pwrModS...` for reactive) will reset always reset value to 0, similar to what turned off relay does. Default value is `nan`, since we can't use 0.0 on account of negative values.   
- fix energy magnitude indexes usage, properly reset in API
- simplify sending pwr and emon Visible flags
- allow to use settings prefixes more, remove snsMinDelta & snsMaxDelta and use relative magnitude in [... cut]
commit 1583548c (2020-07-03): sns: more magnitude settings (#2290)

- introduce zero threshold setting to reset value to 0 below a certain point (resolve #2264)
for example, set pwrPZeroThreshold0 10.0 for active power (or pwrQ... for apparent, pwrModS... for reactive) will reset always reset value to 0, similar to what turned off relay does. Default value is nan, since we can't use 0.0 on account of negative values.
- fix energy magnitude indexes usage, properly reset in API
- simplify sending pwr and emon Visible flags
- allow to use settings prefixes more, remove snsMinDelta & snsMaxDelta and use relative magnitude inde [... cut]
closed 2020-07-03 by mcspr