xoseperez/espurna #2266: Not compatible with SONOFF RE5V1C

sagitt, 2020-05-30:
I have this SONOFF RE5V1C
small relay, 5v power, dry contact.
useful for gate remote, intercom etc.

i flashed it with espruna for Sonoff SV (5v too).. but won't work.. i can see it in my wi-fi networks but can't access to web server page. for now i'm using it with tasmota.

Do you think to support it?

thanks.

[image]

mcspr, 2020-05-30:
Did you use tasmota binary first? Full flash erase is likely required, WiFi might be broken because of the different Core versions used.

If not, can you try changing 
https://github.com/xoseperez/espurna/blob/e26ba9f52ecd60afbbed2ec13d80f4072dad9607/code/platformio.ini#L506
to `env:esp8266-1m-latest-base` and build it yourself?

sagitt, 2020-06-03:
> Did you use tasmota binary first? Full flash erase is likely required, WiFi might be broken because of the different Core versions used.
>
> If not, can you try changing [...]

i erased with esptool and flashed espurna-1.14.1-itead-sonoff-sv.bin. i can see it in wifi list but no configuration page like my other sonoff. i flashed it with last tasmota and works. do you think there is another espruna pre-complied firmware that i can test? i don't know how to self-build. maybe someone can send me a pre-compiled file?

mcspr, 2020-06-04:
What OS, browser do you use?

See
[espurna-issue-2266-itead-sonoff-sv-test.zip](https://github.com/xoseperez/espurna/files/4728457/espurna-issue-2266-itead-sonoff-sv-test.zip)
Erasing still required. Either firmware may misinterpret settings, WiFi might break unexpectedly.