xoseperez/espurna #2270: HDC1080 don't work

jnogues, 2020-06-04:
Hi. I'm compiling the last dev version for a nodemcu. I add suport for HDC1080 but it not works. With another like BMP280 works ok.

Regards.

mcspr, 2020-06-04:
_What_ does not work? It does not build, build does not include the sensor, sensor isn't discovered?

What do debug logs say?

jnogues, 2020-06-04:
When espurna start, here is the serial log

`
[000406] [I2C] Using GPIO04 for SDA and GPIO05 for SCL
[000416] [I2C] Device found at address 0x40
[000423] [I2C] Device found at address 0x76
[000427] [SENSOR] Initializing ANALOG @ TOUT
[000428] [SENSOR]  -> analog:1
[000428] [SENSOR] Initializing BMP280 @ I2C (0x00)
[000438] [I2C] Address 0x76 locked
[000452] [SENSOR]  -> temperature:1
[000452] [SENSOR]  -> humidity:1
[000453] [SENSOR]  -> pressure:1
[000453] [SENSOR] Initializing Dallas @ GPIO10
[000454] [GPIO] GPIO10 locked
[000473] [SENSOR]  -> temperature:2
[000473] [SENSOR] Initializing HDC1080 @ I2C (0x00)
[000474] [I2C] Address 0x40 locked
`
The other sensors works.

mcspr, 2020-06-04:
Right after, there are no temperature:3 and humidity:2? Are there any `[SENSOR] ERROR` shown?

jnogues, 2020-06-04:
No, nothing more about HDC1080. No sensor error

mcspr, 2020-06-04:
Does it work without BMP280 connected?

Can you try modifying
https://github.com/xoseperez/espurna/blob/d9cb35f7814fdb90ca413f7474f0c915a56cbd24/code/espurna/sensors/HDC1080Sensor.h#L130
to
```cpp
_ready = (_chip == HDC1080_CHIP_HDC1080);
```
?

We do seemingly have a bug here:
https://github.com/xoseperez/espurna/blob/d9cb35f7814fdb90ca413f7474f0c915a56cbd24/code/espurna/sensors/HDC1080Sensor.h#L20
https://github.com/xoseperez/espurna/blob/d9cb35f7814fdb90ca413f7474f0c915a56cbd24/code/espurna/sensors/HDC1080Sensor.h#L112-L114
When chip id does not match the constant, sensor it set as both `_ready = true` and `_error = SENSOR_ERROR_UNKNOWN_ID`. So, it does not try again even with error set, because initialization check for readiness first:
https://github.com/xoseperez/espurna/blob/d9cb35f7814fdb90ca413f7474f0c915a56cbd24/code/espurna/sensor.cpp#L2129-L2130

also, cc @vtochq

jnogues, 2020-06-04:
Hi. If I do the change in line 130 to  '_ready = (_chip == HDC1080_CHIP_HDC1080);' there is this error:
'210298] [SENSOR] Initializing HDC1080 @ I2C (0x00)
[210299] [I2C] Address 0x40 locked
[210300] [SENSOR]  -> ERROR 4
'

vtochq, 2020-06-05:
> Quoting http://www.ti.com/lit/ds/symlink/hdc1080.pdf?ts=1591287729001
>
> 8.6 Register Map
> ...
> 0xFB Serial ID device dependent (2 bytes)
> 0xFC Serial ID device dependent (2 bytes)
> 0xFD Serial ID device dependent (1 byte)
> 0xFE Manufacturer ID 0x5449 ID of Texas Instruments
> 0xFF Device ID 0x1050 ID of the device
>
> 8.6.4 Serial Number Registers
> These registers contain a 40bit unique serial number for each individual
> HDC1080.
>
> Device ID is constant, unlike serial. If check is even needed

Hi,

When I made HDC support I think that I will have many HDC devices on one
bus and I will try to identify it by serial.
Actually it's not convenient and very rare implementing.

Max suggests use Device ID, I think something like that:

// Last two bytes of Device and Serial ID
#define HDC1080_CHIP_HDC1080     0x1050

...

// Check device
            i2c_write_uint8(_address, *0xFF*);
            _chip = i2c_read_uint16(_address);
...

unsigned *short* _chip;

Also there are clones of HDC1080 and they may have different Device ID
values. You need to check it by reading and debug output this bytes.

jnogues, 2020-06-05:
Solvedt! Thanks for the help.

regards

jnogues, 2020-06-05:
Perfect! Solved!

PR #2291 (2020-07-03): sns/hdc1080: check device id instead of serial
resolve #2270
- document device ID logic
- log when we receive unexpected device ID
- don't fall-thought to '_ready=true' not ready
closed 2020-07-03 by mcspr