xoseperez/espurna #2271: dev branch only publish temperature by mqtt the first time

jnogues, 2020-06-05:
With the latest dev version, I have tested different temperature sensors, ds18b20, bmp280 and hdc1080, on a nodemcu.

They all measure the temperature well and it is displayed on the web gui. By mqtt only the temperature is published the first time. In the rest of the publications, in json or individually no longer.

The rest of the measurements, humidity and / or pressure are published and are updated on the web.

Regards.

mcspr, 2020-06-05:
Not seeing this with DummySensor. 

Something with DS18B20?
https://gitter.im/tinkerman-cat/espurna?at=5ea6d54b1e3d5e2063493c6a

jnogues, 2020-06-06:
With DS18B20 I have the same behaviour. :(

jnogues, 2020-06-06:
Now I add the DummySensor. I have 4 sensors. DS18b20, bmp280, hdc1080 and dummy.
The first publish OK. In the next, the dummy sensor ok, but the others temperatures aren't publish. Can we see the same in serial log:
 First publish:

```
[060011] [MAIN] System OK
[060177] [MQTT] Sending /espurna/ESPURNA-49C55D/analog => 460 (PID 1)
[060178] [MQTT] Sending /espurna/ESPURNA-49C55D/temperature/0 => 25.795 (PID 1)
[060180] [MQTT] Sending /espurna/ESPURNA-49C55D/humidity/0 => 50 (PID 1)
[060186] [MQTT] Sending /espurna/ESPURNA-49C55D/pressure/0 => 1003.3043 (PID 1)
[060193] [MQTT] Sending /espurna/ESPURNA-49C55D/temperature/1 => 25.59 (PID 1)
[060200] [MQTT] Sending /espurna/ESPURNA-49C55D/temperature/2 => 26.8 (PID 1)
[060206] [MQTT] Sending /espurna/ESPURNA-49C55D/humidity/1 => 56 (PID 1)
[060213] [MQTT] Sending /espurna/ESPURNA-49C55D/temperature/3 => 30.5 (PID 1)
[060223] [MQTT] Sending /espurna/ESPURNA-49C55D/humidity/2 => 56 (PID 1)
[060230] [MQTT] Sending /espurna/ESPURNA-49C55D/pressure/1 => 1005.50 (PID 1)
[060235] [MQTT] Sending /espurna/ESPURNA-49C55D/lux => 6 (PID 1)

```
Second publish, /temperature/0 and /temperature/1 and temperature/2, missing:

```
[180274] [MQTT] Sending /espurna/ESPURNA-49C55D/analog => 460 (PID 1)
[180276] [MQTT] Sending /espurna/ESPURNA-49C55D/humidity/0 => 50 (PID 1)
[180277] [MQTT] Sending /espurna/ESPURNA-49C55D/pressure/0 => 1003.3100 (PID 1)
[180284] [MQTT] Sending /espurna/ESPURNA-49C55D/humidity/1 => 56 (PID 1)
[180289] [MQTT] Sending /espurna/ESPURNA-49C55D/temperature/3 => 10.0 (PID 1)
[180296] [MQTT] Sending /espurna/ESPURNA-49C55D/humidity/2 => 75 (PID 1)
[180302] [MQTT] Sending /espurna/ESPURNA-49C55D/pressure/1 => 995.67 (PID 1)
[180309] [MQTT] Sending /espurna/ESPURNA-49C55D/lux => 25 (PID 1)
```

mcspr, 2020-06-06:
Do you have `TEMPERATURE_MIN_CHANGE` set?
https://github.com/xoseperez/espurna/blob/d9cb35f7814fdb90ca413f7474f0c915a56cbd24/code/espurna/sensor.cpp#L2217-L2220
https://github.com/xoseperez/espurna/blob/d9cb35f7814fdb90ca413f7474f0c915a56cbd24/code/espurna/sensor.cpp#L2403
https://github.com/xoseperez/espurna/blob/d9cb35f7814fdb90ca413f7474f0c915a56cbd24/code/espurna/sensor.cpp#L2694-L2695

jnogues, 2020-06-06:
I have the 3 files like you say me, but where have I set `TEMPERATURE_MIN_CHANGE` ?

jnogues, 2020-06-06:
Solved!
Max, sorry for my incompetence, I had a definition of `TEMPERATURE_MIN_CHANGE` to 1.0 without realizing it.

Sorry for wasting your time!!!!

Regards!

closed 2020-06-07 by mcspr