xoseperez/espurna #2272: Firmware for NEDIS [WIFIP310FWT] Wi-Fi Smart Extension Socket | 3x Schuko Type F | 4x USB | 16 A

mindtripper, 2020-06-05:
Is there any precompiled firmware for Nedis Wi-Fi Smart Extension Socket | 3x Schuko Type F | 4x USB | 16 A available? I've looked, but haven't found anything.
Maybe there is a generic FW or FW for another device that is compatible?

Brand: Nedis
Vendor Part Number: WIFIP310FWT
EAN number: 5412810271795

[Link to product page](https://nedis.com/en-us/product/smart-living/smart-home/energy/550672299/wi-fi-smart-extension-socket-3x-schuko-type-f-4x-usb-16-a)

mcspr, 2020-06-05:
Tasmota Device Templates Repository project maintains an excellent device database:
https://templates.blakadder.com/nedis_WIFIP310FWT.html
Which we could use to reference GPIO settings, see on the right

mindtripper, 2020-06-05:
Thanks again for the fast response and feedback. I'm impressed!

Yeah, I've seen that site and considered putting Tasmota on the device. But I'd prefer having a homogenous environment using ESPURNA. Can this device be added to the nightly builds list?

mcspr, 2020-06-05:
What I meant was to c/p GPIO numbers from the page :) I'll add the profile, just need to fix local git setup.

For example, see:
https://github.com/xoseperez/espurna/blob/d9cb35f7814fdb90ca413f7474f0c915a56cbd24/code/espurna/config/hardware.h#L2718-L2752
Button, relays numbers are similar. (RELAY1_PIN, BUTTON1_PIN etc. flags)

mindtripper, 2020-06-06:
Being very curious I went ahead with tuya-covert and put Zhilde ZLD-EU44-W 1.15.0 daily build firmware on the device. As you say, it has similar GPIO setup and it works, but having a dedicated firmware would be better.

Looking at Zhilde ZLD-EU44-W firmware on [NEDIS [WIFIP310FWT]](https://cdn.nedis.com/images/products_350/WIFIP310FWT_P50.JPG)
This is how this firmware works based on the picture above (top to bottom):
Switch #4
USB
Switch #0
Socket 1 
Switch #1
Socket 2
Switch #3
Socket 3
_Switch #2
N/A_

I'll test the NEDIS build if you decide to add it 🥇

mcspr, 2020-06-07:
I forgot what ANALOG flag means in Tasmota templates though. Something to do with ADC, but what exactly...

closed 2020-06-07 by mcspr

mcspr, 2020-06-07:
nedis-wifip310fwt-sensor is the 2nd bin, with ANALOG_SUPPORT sensor enabled

PR #2275 (2020-06-07): hardware: Add support for Nedis WIFIP310FWT
resolve #2272