xoseperez/espurna #2273: Teckin SP22 web interface not working after firmware upgrade

javalc6, 2020-06-06:
Hello, I've successfully installed the plain espurna.bin firmware on Teckin SP22 device.
I connected to the device on 192.168.4.1 to access web interface, it worked with plain firmware.
I've updated the firmware via web interface, but now I'm no longer able to connect to the device web interface.
The problem is that when I access 192.168.4.1:80, I get a chunk of bytes, that is not understood by any browser I've tried.

Faulty [firmware=](https://github.com/mcspr/espurna-nightly-builder/releases/download/20200601/espurna-1.15.0-dev.nightly20200601+gitd9cb35f7-teckin-sp22-v14.bin)

_Device Information_
[205655] [MAIN] ESPURNA 1.15.0-dev (d9cb35f7)
[205655] [MAIN] xose.perez@gmail.com
[205655] [MAIN] http://tinkerman.cat
[205655] [MAIN] CPU chip ID: 0xEF92A7
[205655] [MAIN] CPU frequency: 80 MHz
[205655] [MAIN] SDK version: 1.5.3(aec24ac9)
[205656] [MAIN] Core version: 2.3.0
[205656] [MAIN] Core revision: 9826c6d
[205656] [MAIN] Build time: 1590978514
[205656]
[205656] [MAIN] Flash chip ID: 0x144051
[205657] [MAIN] Flash speed: 40000000 Hz
[205657] [MAIN] Flash mode: DOUT
[205657]
[205658] [MAIN] Flash size (CHIP)   :  1048576 bytes /  256 sectors (   0 to  255)
[205658] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[205658] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[205659] [MAIN] Firmware size       :   524128 bytes /  128 sectors (   1 to  128)
[205659] [MAIN] Max OTA size        :   499712 bytes /  122 sectors ( 129 to  250)
[205659] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[205660] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[205660]
[205660] [MAIN] EEPROM sectors: 251, 250
[205660] [MAIN] EEPROM current: 250
[205660]
[205661] [MAIN] EEPROM:  4096 bytes initially |    69 bytes used ( 1%) |  4027 bytes free (98%)
[205661] [MAIN] Heap  : 29072 bytes initially | 23016 bytes used (79%) |  6056 bytes free (20%)
[205662] [MAIN] Stack :  4096 bytes initially |  1584 bytes used (38%) |  2512 bytes free (61%)
[205663]
[205663] [MAIN] Boot version: 31
[205663] [MAIN] Boot mode: 1
[205663] [MAIN] Last reset reason: Software/System restart
[205664] [MAIN] Last reset info: Fatal exception:0 flag:4 (SOFT_RESTART) epc1:0x00000000 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
[205664]
[205664] [MAIN] Board: TECKIN_SP22_V14
[205665] [MAIN] Support: ALEXA API BROKER BUTTON DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT LED MDNS_SERVER MQTT NTP SCHEDULER SENSOR TELNET TERMINAL THINGSPEAK WEB
[205665] [MAIN] OTA: ARDUINO ASYNCTCP WEB
[205665] [MAIN] Sensors: HLW8012
[205665] [MAIN] WebUI image: SENSOR
[205665]
[205905] [MAIN] Firmware MD5: 765aaa6a3b13914eef01d6d553faacb9
[205908] [MAIN] Power: 3182 mV
[205909] [MAIN] Power saving delay value: 10 ms
[205909]

mcspr, 2020-06-06:
What OS and browser are you using?

Maybe duplicate of #2256, debugging is pain. As noted in the other issue, power-cycle does help

javalc6, 2020-06-06:
I've tried Chrome, Firefox and IE from Windows 10; Chrome from Android 9; always same result, power-cycle (off-on) tried several times, no improvements.
On the other hand I can connect via telnet, but it is would be better to have ssh instead of telnet.

javalc6, 2020-06-06:
I've just changed configuration of the device using command `curl -u admin:fibonacci http://192.168.4.1/config` in order to set-up wifi access via my home network.
I can access the web gui interface using the IP address assigned by my wifi router.
So the problem is related only to the access via SoftAP.

closed 2020-06-06 by javalc6

mcspr, 2020-06-07:
We can try switching SDK Core versions on fresh units, as those are the only ones that seem to break this way? (which might be a good idea overall, if stock firmware causes this weird transition)

I wonder if client order matters? We did have #1660 because of Android captive portal detection being to eager to flood us with requests completely shutting down the webserver until reset or even more requests finally rebooting because malloc() stops working.

Re `Heap : 29072 bytes initially | 23016 bytes used (79%) | 6056 bytes free (20%)`, 6K is pretty low, but should've worked at least loading the index page.