xoseperez/espurna #2277: PZEM004T, V3.0 serial and libraries.

davebuk, 2020-06-14:
I've just ordered a V3 PZEM004T and plan to use it with a wemos d1. I was going to use the TX/RX pins as the entry in `sensors.h` says software serial is not working. I guess that actually isn't the case and I can set software serial support and choose my own GPIO pins?

_Originally posted by @davebuk in https://github.com/xoseperez/espurna/issues/1583#issuecomment-643665930_

As well as soft serial and hard serial settings, do the libraries that espurna uses work with the V3.0 boards?

mcspr, 2020-06-14:
v3 is something different from the original pzem

https://github.com/mandulaj/PZEM-004T-v30 seems to implement the new active protocol, we can expose it as a sensor

davebuk, 2020-06-14:
That would be great if you could. I believe it has a couple more magnitudes, `frequency` and maybe  `power factor`, so they will need adding. I don't have the hardware yet, it'll be a week or so, but I can test it if you are happy to create the code?

mcspr, 2020-06-14:
Sure, if you also don't mind testing a different settings backend (which might break) at the same time :)

It just a matter of
- adding a new sensor id, class, configuration entries (PZEM004TV3_SUPPORT, etc.) and loader in the sensor.cpp
- writing sensor class to create library's object on begin()
- adding library.updateValues() in tick(), reading values after that. 
- disabling energy saver same as the original code, we trust pzem device to manage it.
- returning stored values on value()

Power factor is an existing measurement type, not sure what frequency needs to be here as we don't have any sensor outputting that.

davebuk, 2020-06-15:
Yep, I can test. It'll be connected to a wemos D1. I plan to add a 433MHz TX/RX to the board as well when it's up and running.

I found a page here that maybe of some use? There is a section with Arduino code examples.
https://innovatorsguru.com/pzem-004t-v3/

davebuk, 2020-06-19:
Hardware has arrived. I'll look to connect it all together over the weekend ready for some testing. It's the 100A version if that makes a difference to the expected output.

mcspr, 2020-06-20:
Will add PR some time today then, what I tried with dummy data based on the brief 'datasheet' description of the protocol seemed to work OK (and at least Hardware Serial version).
I did not use the library though, but I guess if that does not work you can try `examples/` from there to understand any possible differences.

100A only changes some 'default' values from the sensor, should work fine from the software side.
And I am not really sure what 'alarm' setting means (setAlarm() in the library, also see the .pdf), will it beep or something?

davebuk, 2020-06-20:
I'll report testing in #2283 

Re: 'alarm', from the data sheet I have with the [device](https://innovatorsguru.com/wp-content/uploads/2019/06/PZEM-004T-V3.0-Datasheet-User-Manual.pdf) it says:

"1.7 Over power alarm
Active power threshold can be set, when the measured active power exceeds the threshold, it
can alarm"

You may already know this but the document shows `Register address 0x0009 is Alarm status (0xFFFF - alarm, 0x0000 - alarm)`. The wattage trigger level can be set and when reached, the `alarm` register is set. I guess this could report as 0 or 1 to mqtt via one of the settings?

closed 2020-07-03 by mcspr

mcspr, 2020-07-03:
> You may already know this but the document shows Register address 0x0009 is Alarm status (0xFFFF - alarm, 0x0000 - alarm). The wattage trigger level can be set and when reached, the alarm register is set. I guess this could report as 0 or 1 to mqtt via one of the settings?

Yeah. But... why. When we receive data packet, we receive all at once and already have power value and can calculate things ourselves. I wonder if this is something related to the display module included in the other versions

davebuk, 2020-07-03:
Agree ref the alarm status. There doesn't seem to be an immediate use case with espurna. Maybe if the sensor unit is running as a stand alone module, without input into an esp8266 style device, it would be of some use like the display module you suggested.

davebuk, 2020-07-03:
I hope to try and build my "production" hardware over the weekend and connect to my house supply. It should then integrate into openHAB.

davebuk, 2020-07-07:
I now have the hardware connected and the sensor works really well. Great work.

[image]

dpeddi, 2020-07-13:
@davebuk do you have a ttl converter between the wemos and the PZE? Please could you share the schematic you used?

davebuk, 2020-07-13:
@dpeddi I don't have anything between the two devices. I have 5VDC, GND, Tx and Rx connected to the Wemos D1 mini. As long as have you have Tx/Rx the correct way round on the GPIOs you have you have configured in the espurna build, the firmware does all the serial communications.

PR #2283 (2020-07-03): Sensor: Add PZEM004T V3.0
@davebuk 

Fix #2277 
Add new V3.0 version of PZEM004T, not tested. Default to factory default address 0xf8, configurable through `pzemv30Addr` 8-bit number
Add 'frequency' measurement & 'hz' unit
Add `pz.address` that configures address
Implemented energy reset

Right now we only support a single device. No way to test this properly (mostly, how exactly we need to connect this stuff), so left as not implemented.