xoseperez/espurna #2279: flashing to alternative firmware doesn't work

craiggenner, 2020-06-16:
**Bug description**
I think the description is clear, but just in case when I try and flash to alternative firmware (tasmota in this case) I'm left with a device that is in a funny state. Some times the device is left in a state that is only fixed by erasing the flash and then reflashing the firmware onto the device (no wifi AP, no connection to exiting wifi). Other times it just stops the web interface and I've had to use the telnet interface to reset and get back in.

**Steps to reproduce**

1. Upload lastest tasmota firmware using web interface
1. Wait for restart  

**Expected behavior**
Start with the alternative firmware

I think what's happening here is a combination of the new and old firmware and settings from the old firmware both existing at the same time on the device. To resolve this I think a "install firmware and erase flash" option is required. Maybe a custom espurna for this purpose.

**Screenshots**
None

**Device information**
* espurna-1.14.1-itead-sonoff-basic.bin
* OTA flash mode
* Sonoff basic

(also seen with 2Nice/gosuna UPC111)

**Tools used**
* esptool for recovery flashing

**Additional context**
I understand that it's not in the best interest of the project to help use alternative firmware, not sure how much response/help I'll get here. About the only positive view from this is the perception that espurna is at fault for not flashing the new firmware, but I suspect most people running espurna are clever enough to understand this.

mcspr, 2020-06-16:
> I understand that it's not in the best interest of the project to help use alternative firmware, not sure how much response/help I'll get here. About the only positive view from this is the [...]

"funny state" does not really describe things clearly. Do you mean that such device is running Tasmota but not being able to connect to anything / start AP? Do you have access to such 'broken' device's serial and can you type `reset 3` and report the result?

Or, are you simply unable to upgrade to anything else via OTA?

craiggenner, 2020-06-16:
> > I understand that it's not in the best interest of the project to help use alternative firmware, not sure how much response/help I'll get here. About the only positive view from this is the perception that espurna is at fault for not flashing the new firmware, but I suspect most people running espurna are clever enough to understand this.
> 
> "funny state" does not really describe things clearly. Do you mean that such device is running Tasmota but not being able to connect to anything / start AP? Do you have access to such 'broken' device's serial and can you type `reset 3` and report the result?
> 
> Or, are you simply unable to upgrade to anything else via OTA and still running Espurna?
Thanks for the response

I've seen it in a state of:
* still running espurna where the new firmware appears to have been ignored (not very often)
* still running espurna but broken (the web interface doesn't start and I have to use telnet)
  * at this stage it either connects back to my wifi or I see the espurna AP
* in a state where I don't know what it's running (doesn't connect to my Wifi or create an AP)

To be honest it appears random which way it goes.

I've tried to get access to the serial, but without much luck.

mcspr, 2020-06-16:
Regarding the first state, can you try `erase.config` command in telnet / webui (when it works...)? It will have a similar effect as `reset 3` in Tasmota, note that it will reboot the device immediately after entering.

Are you trying to upgrade with -minimal binary? See
https://github.com/xoseperez/espurna/issues/1663#issuecomment-592567802
> FTR, never use tasmota-minimal.bin as the first flash of Tasmota. It depends on config data from previous Tasmota install, and will only work as an intermediary step going from one Tasmota binary to another. This two-step procedure is to enable binaries with a size over half the available flash space within 1M.

craiggenner, 2020-06-18:
I've tried full regular tasmota and the -minimal version without luck.

From the linked comment I can see why the device gets into the last state (no wifi or AP) so I'll avoid using that firmware from now on.

craiggenner, 2020-06-20:
I've now got this working through the following process:
* flash with `espurna-x.x.x-espurna-core.bin`
  * I was consistent in using the same core version as what was already running, when using a version that **wasn't** the same as what I was running the 'upgrade' to the core version wasn't always a success.
* make the `tasmota.bin` file available on the local network over http (I only tried http, not https)
  * Ran `python -m SimpleHTTPServer` in the same directory as the bin file
* connected via telnet to the device and issued the `ota <url>` command, eg:
  * telnet 192.168.0.3
  * ota http://192.168.0.2:8080/tasmota.bin
* You know if this is working as you see the python http server reporting the http request
* Wait a while and keep checking for the tasmota wifi to appear, when this happens you are done

Note the following - Some devices take longer than others to switch to the `espurna-core`, then to request the `tamsota.bin` file, and for the tasmota wifi to appear. The length of time does appear consistent over the hardware types though (sonoff basic, teckin plug, 2nice plug, etc)

Also note that the telnet commands might not give you any response at all, this appears to be a client issue though.

Thanks for your hints on this. I hope someone finds the notes above of use.

I'll close this issue now.

closed 2020-06-20 by craiggenner

mcspr, 2020-07-08:
Re: "Also note that the telnet commands might not give you any response at all, this appears to be a client issue though."
https://github.com/xoseperez/espurna/blob/2be8d106014d824ce362a33ca663a8429910c52b/code/espurna/config/hardware.h#L60
Debug output is disabled, but not input. I might need to change that...