xoseperez/espurna #2281: Support for Litesun LA-WF7 ?

zuberio, 2020-06-18:
CE Smart Home Wi-Fi Smart Plug with Energy Monitoring, 2-pack sold by Costco in Canada for $20.
https://www.costco.ca/ce-smart-home-wi-fi-smart-plug-with-energy-monitoring%2C-2-pack.product.100476838.html

Tuya Device, compatible with Tuya-Convert (if you've never paired with the app).

code/espurna/config/arduino.h:
`//#define LITESUN_LA-WF7`

code/espurna/config/hardware.h:
```
// -----------------------------------------------------------------------------
// LITESUN LA-WF7
// https://www.costco.ca/ce-smart-home-wi-fi-smart-plug-with-energy-monitoring%2C-2-pack.product.100476838.html
// -----------------------------------------------------------------------------

#elif defined(LITESUN_LA-WF7)
    // Info
    #define MANUFACTURER                "LITESUN"
    #define DEVICE                      "LA-WF7"

    // Buttons
    #define BUTTON1_PIN                 3
    #define BUTTON1_MODE                BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON1_RELAY               1

    // Relays
    #define RELAY1_PIN                  14
    #define RELAY1_TYPE                 RELAY_TYPE_NORMAL

    // LEDs
    #define LED1_PIN                    13
    #define LED1_PIN_INVERSE            1
    #define LED1_MODE                   LED_MODE_FINDME_WIFI

    // HJL01 / BL0937
    #ifndef HLW8012_SUPPORT
    #define HLW8012_SUPPORT             1
    #endif
    #define HLW8012_SEL_PIN             12
    #define HLW8012_CF1_PIN             5
    #define HLW8012_CF_PIN              4

    #define HLW8012_SEL_CURRENT         LOW
    #define HLW8012_CURRENT_RATIO       21683
    #define HLW8012_VOLTAGE_RATIO       262278
    #define HLW8012_POWER_RATIO         2714640
    #define HLW8012_INTERRUPT_ON        FALLING
```

code/espurna/migrate.ino:
```
        #elif defined(LITESUN_LA-WF7)

            setSetting("board", 79);
            setSetting("ledGPIO", 0, 13);
            setSetting("ledLogic", 0, 1);
            setSetting("ledMode", 0, LED_MODE_FINDME_WIFI);
            setSetting("btnGPIO", 0, 3);
            setSetting("btnRelay", 0, 0);
            setSetting("relayGPIO", 0, 14);
            setSetting("relayType", 0, RELAY_TYPE_NORMAL);
            setSetting("selGPIO", 12);
            setSetting("cf1GPIO", 5);
            setSetting("cfGPIO", 4);
            setSetting("pwrRatioC", 21683);
            setSetting("pwrRatioV", 262278);
            setSetting("pwrRatioP", 2714640);
            setSetting("hlwSelC", LOW);
            setSetting("hlwIntM", FALLING);
```

code/platformio.ini:
```
[env:litesun-la-wf7]
extends = env:esp8266-1m-base
src_build_flags = -DLITESUN_LA_WF7

```

[image]
[image]
[image]
[image]

IF you made the mistake (like me) of pairing these devices with the CE Smart Home app & also 'upgrading' the device WiFi firmware via said app (like me) - then Tuya-Convert won't work. So I measured out where to drill into the device casing in order to reach the TYWE2S: GPIO0, 3V3, GND, RX and TX pins for physical flashing.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

mcspr, 2020-06-18:
Thanks!

Based on https://templates.blakadder.com/ce_smart_home_LA-WF7.html, we already have something like it via SPHX_V23. Ratios for power measurement are different though.
https://github.com/xoseperez/espurna/blob/7aec0c57eb18070e08082d3feebcabea1459ab62/code/espurna/config/hardware.h#L3124-L3163
You might want to follow the button config with pull-up and disable serial output, this may cause some minor issues.

zuberio, 2020-06-18:
Oh! Thank you! I based my config on the BLITZWOLF_BWSHPX, instead of the BLITZWOLF_BWSHPX_V23 - silly me!

wipeout666, 2020-11-16:
Thanks @zuberio for this post, was able to recover my failed OTA flashing of my LA-WF7 without blowing up the whole case: 

[image]

zuberio, 2020-11-16:
@wipeout666 glad I could help!

mcspr, 2020-11-16:
Closing via https://github.com/xoseperez/espurna/wiki/Hardware-Litesun-LA-WF7
I edited the https://github.com/xoseperez/espurna/wiki/Hardware back in June but forgot to commit images as well... Thanks!

closed 2020-11-16 by mcspr

LexxM3, 2021-02-24:
I just wanted to say thank you for this by describing a technique that allows these plugs to be opened without destroying the case and allowing "as good as factory" resealing after flashing.

See pics attached.  The basic process (and one that can be used on many glued products) is to use a heat gun and (sacrificial) plastic guitar picks.  Hold the heat gun around 1ft from the plug so as not to burn the plastic and have patience.  Heat around the sides and on the top.  Find a slight gap on one of the two long sides and insert a fine guitar pick as a wedge.  Slide the guitar picks firmly through the gap, gradually opening up more and more of the cover.  The corners are the hardest and the key to success -- when you've slid through all 4 corners, you know it's going to work.  Reheat if needed. Once you can slide the picks all the way around, if it hasn't popped off yet, just keep on going around and apply firm pressure down -- it will pop off within a couple of passes.  I used two thicknesses of guitar picks: 0.58mm as a starter and 0.73mm as a finisher.  Once the cover is off, the PCB is just pushed out by applying pressure to the prongs against a hard surface.  Once reprogramming is finished, a little glue seals it all up.

**OBVIOUSLY USE EXTREME ELECTRICAL SAFETY CAUTION WHEN WORKING ON THE PLUGGED-IN BOARD WITHOUT THE CASE!!!**

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

zuberio, 2021-02-24:
Nice! Do you also find that the button doesn't feel the same once the cover has been popped off?

LexxM3, 2021-02-24:
> Nice! Do you also find that the button doesn't feel the same once the cover has been popped off?

That’s a good question. First, there is a trick to putting the plastic plunger back in: use tape on top of the cover to hold it in place against the back when installing the cover back on. Second, yes, it does feel a little different, just a little less travel, but still works fine.