xoseperez/espurna #2285: No NTP time sync with 1.15.0-dev

davebuk, 2020-06-23:
I have two devices that have been up for 23 days running 1.15.0-dev, but they have never got a time using NTP. All my other devices running 1.14.1 are fine.

I've tried changing the NTP server but that hasn't helped.

Are there other settings I can check?

mcspr, 2020-06-27:
What does terminal `get ntpServer` show?
Does it work on a different network?

davebuk, 2020-06-27:
I have two devices running 1.15.0-dev on the same network. Neither have had a time for over 20 days. I've just checked and for some reason, one device rebooted an hour ago with a hardware watchdog reset. That one has now synced.

The second device was up for 27 days without sync. I've just power cycled the 240V rather than just a software reboot and with 12 mins uptime, it still hadn't got a sync.
Debug commands give:
> ntpServer => "pool.ntp.org"
and
[NTP] Not synced

I've just connected the device to my mobile as a hotspot. Left it a few minutes and reconnected back to the network. It has now synced!

Can I force an ntp sync to check it can talk to the NTP server?

mcspr, 2020-06-27:
Have you set it manually from ui / config upload? If not, does it come back after `del ntpServer` & reboot? 
What does `host pool.ntp.org` show?

davebuk, 2020-06-27:
I haven't touched any NTP settings, just default.
Debug
[DNS] pool.ntp.org not found

mcspr, 2020-06-27:
We do have pool.ntp.org as default, so that is kind of expected to see in the ui, what I was thinking about is override from DHCP which also sets `ntpServer` to something. If `get ntpServer` now does not show anything, I'd guess network configuration is weird somehow? Do any other NTP servers work though, time.nist.gov for example? Or something local via IP, most routers do have sntp server for example

`wifi` command also shows the DNS server used, maybe the problem is there?
Something like `nslookup pool.ntp.org <whatever-ip-DNS-server-is>` on a PC should not work too with that server?

davebuk, 2020-06-28:
Think I've found the issue. I'm using static IP addresses for all my devices. I have about 20 devices and found one running 1.14.1 that also hadn't synced.

On most devices I'd set 8.8.8.8 for the DNS server, but the devices not syncing I'd use my router address. Changing them to 8.8.8.8 now works.

Thanks for your help.

closed 2020-06-28 by davebuk