xoseperez/espurna #2287: ota.py errors when searching for espurna devices

CrispinP, 2020-06-24:
Hi,

Getting error when trying to run ota to search for devices to update.

It does not work with python2.7 or 3.7

[image]

What am I doing wrong?

mcspr, 2020-06-24:
zeroconf stuff changed, perhaps?
does `pip install --user zeroconf==0.24.5` fix this? (or 0.24.0, see requirements.txt)

mcspr, 2020-06-25:
Quoting changelog at https://github.com/jstasiak/python-zeroconf

> 0.27.0
> ...
> Removed previously deprecated ServiceInfo address constructor parameter and property

I also found this patch https://github.com/home-assistant-libs/pychromecast/commit/6c07e0327fd87de3b1e9511c843bc6ef166dc5b8#diff-c86c725f2be6a5290140ac27b5c96835, which is something this when translating it for our code:
```diff
diff --git a/code/ota.py b/code/ota.py
index 3541fcbb..c7750146 100755
--- a/code/ota.py
+++ b/code/ota.py
@@ -113,9 +113,11 @@ class Listener:
             return

         hostname = info.server.split(".")[0]
+        addresses = info.parsed_addresses()
+
         device = {
             "hostname": hostname.upper(),
-            "ip": socket.inet_ntoa(info.address),
+            "ip": addresses[0] if addresses else info.host,
             "mac": "",
             "app_name": "",
             "app_version": "",
```

PR #2299 (2020-07-08): ota.py: don't use deprecated zeroconf attrs
Also bump zeroconf to the version used by PIO
fix #2287
closed 2020-07-08 by mcspr