xoseperez/espurna #2289: From 1.14.1 to 1.15.1.dev Sonoff Dual R2 stopped to work

rmcbc, 2020-06-27:
**Bug description**
I migrated from 1.14.1 to 1.15.1.dev and only one of the buttons work, get back to 1.14.1 and all works ok.

In my case BUTTON2 work in both versions and BUTTON1 only work on 1.14.1

I change from pin 0 to pin 3 because pin 0 can be used to enter in programming mode.

Button configuration in my custom.h :
'
    // Buttons
    #define BUTTON1_PIN         3       // Button RX on header
    //#define BUTTON1_PIN         0       // Button 0 on header
    #define BUTTON2_PIN         9       // Button 1 on header
    #define BUTTON3_PIN         10      // Physical button
    #define BUTTON1_RELAY       1
    #define BUTTON2_RELAY       2
    #define BUTTON3_RELAY       1
    #define BUTTON1_CONFIG       BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH | BUTTON_SET_PULLUP
    #define BUTTON2_CONFIG       BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH | BUTTON_SET_PULLUP
    #define BUTTON3_CONFIG       BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH

    #define BUTTON1_PRESS       BUTTON_MODE_TOGGLE
    #define BUTTON1_CLICK       BUTTON_MODE_NONE
    #define BUTTON1_DBLCLICK    BUTTON_MODE_NONE
    #define BUTTON1_LNGCLICK    BUTTON_MODE_NONE
    #define BUTTON1_LNGLNGCLICK BUTTON_MODE_NONE

    #define BUTTON2_PRESS       BUTTON_MODE_TOGGLE
    #define BUTTON2_CLICK       BUTTON_MODE_NONE
    #define BUTTON2_DBLCLICK    BUTTON_MODE_NONE
    #define BUTTON2_LNGCLICK    BUTTON_MODE_NONE
    #define BUTTON2_LNGLNGCLICK BUTTON_MODE_NONE

    #define BUTTON1_PRESS       BUTTON_MODE_TOGGLE
    #define BUTTON1_CLICK       BUTTON_MODE_NONE
    #define BUTTON1_DBLCLICK    BUTTON_MODE_NONE
    #define BUTTON1_LNGCLICK    BUTTON_MODE_NONE
    #define BUTTON1_LNGLNGCLICK BUTTON_MODE_NONE  
'

mcspr, 2020-06-27:
Can you post the whole config that is used for the device in question?
Are you using BUTTON1_CONFIG flag for both versions?

rmcbc, 2020-06-27:
[custom.h.zip](https://github.com/xoseperez/espurna/files/4841411/custom.h.zip)

> Can you post the whole config that is used for the device in question? yes

> Are you using BUTTON1_CONFIG flag for both versions? No, 1.14.1 BUTTON1_MODE and 1.15.1.dev BUTTON1_CONFIG

mcspr, 2020-06-28:
`BUTTON_REPEAT_DELAY` seems to have an effect on single clicks, it takes 10 seconds for "click" event to appear.
Not really sure as to why though, will look into it.

mcspr, 2020-07-04:
Not really seeing any difference between versions, I'd look at delays setup / compare settings with stock hardware profile.

Perhaps there was some confusion in naming, as previously this setting was referring to the Double-click only.
```
[time from BUTTON_REPEAT_DELAY passes...]
click  .... click ....
```
-> double click after delay window ends

```
[time from BUTTON_REPEAT_DELAY passes...]
click .... (and nothing more)
```
-> click happens only after delay window ends

closed 2020-09-11 by stale[bot]