xoseperez/espurna #2292: simple voltage (12v) monitoring with nodemcu

djsomi, 2020-07-03:
Hi All,

I need to build a simple  monitoring for my 12V bus system with NodeMCU. (or any other simple board)
Need to collect voltage data, with influx support and visualize with grafana. (I already using these tools for temp and hum monitoring, they are working fine)

I know we can measure analog data on NodeMCU A0 pin with some resistors. Is there someone who already implemented this with this FW? Any suggestion and idea is welcome how to build a custom fw for this purpose.

Many Thanks
Zoli

mcspr, 2020-07-03:
Have you tried just using `ANALOG_SUPPORT` to read the value? Note of the factor setting, which will multiply the 0...1023 integer accordingly.
There is also ad-hoc `EMON_ANALOG_SUPPORT`, which needs the correct voltage setting and will try to emulate generic energy monitor sensors by calculating power and resulting energy values.

For configuration flags, see https://github.com/xoseperez/espurna/blob/dev/code/espurna/config/sensors.h

closed 2020-09-11 by stale[bot]

djsomi, 2020-10-05:
Any experiences on INA voltage sensors?