xoseperez/espurna #2296: Does espurna support 433MHz SRX882 receiver?

davebuk, 2020-07-05:
I was looking to add a 433MHz SRX882 receiver to a wemos D1 mini to pick up some two code, RF door sensors and send the open/closed signal to openHAB. I have one running already with 'OpenMQTTGateway' firmware but wanted to look at keeping as many devices as possible running espurna.

I have added the following build flags and connected the receiver to the defined GPIO. Espurna shows 8 dummy switches and and the option to 'LEARN' codes for each switch. When I trigger an RF sensor, I don't see any codes appear. Are there more settings required to get the receiver working?

```
#define RELAY_PROVIDER          RELAY_PROVIDER_RFBRIDGE

// Number of virtual switches
#ifndef DUMMY_RELAY_COUNT
#define DUMMY_RELAY_COUNT   8
#endif	

// RF Support
#define RF_SUPPORT              1
#define RFB_DIRECT              1
#define RFB_RX_PIN              5 // D1
```

mcspr, 2020-07-06:
Depends on the type of RF used by the OpenMqttGateway. If a simple rc-switch, we can try this:
```diff
diff --git a/code/platformio.ini b/code/platformio.ini
index 3ed747d1..90271443 100644
--- a/code/platformio.ini
+++ b/code/platformio.ini
@@ -156,7 +156,7 @@ lib_deps =
     OneWire
     PZEM004T
     PubSubClient
-    rc-switch
+    https://github.com/1technophile/rc-switch#8fc6d06c
     https://github.com/LowPowerLab/RFM69#7008d57a
     https://github.com/xoseperez/rpnlib.git#0.3.0
     https://github.com/xoseperez/Time
```
Codes *will* be incorrect length though if protocol has 64 bit data, as old rc-switch (and what we support) only had 32 bits

davebuk, 2020-07-06:
I deleted `rc-switch` and added `https://github.com/1technophile/rc-switch#8fc6d06c` but no difference. Is there a way in the debug screen to see if the RF device is sending any data to espurna?

mcspr, 2020-07-07:
For basic protocol and remotes like this one https://sonoff.tech/product/accessories/rm433
```
[021381] [RF] Received code: 00EA260F
[021381] [RF] Action 0xA4
[021382] [RF] Received message 'C00101861800EA260F'
[021382] [RF] Trying to match code EA260F
```

This still assumes that OpenMqttGateway env used rc-switch too, as seen here in the lib_deps:
https://github.com/1technophile/OpenMQTTGateway/blob/development/platformio.ini
Meaning, env like nodemcuv2-rf

davebuk, 2020-07-07:
I'm using the following to build the firmware in platformIO

```
[env:nodemcuv2-ir-rf]
platform = ${com.esp8266_platform}
board = nodemcuv2
lib_deps =
  ${com-esp.lib_deps}
  ${libraries.irremoteesp}
  ${libraries.rc-switch}
  ${libraries.esp8266_mdns}
build_flags = 
  ${com-esp.build_flags}
  '-DZgatewayIR="IR"'
  '-DZgatewayRF="RF"'
  '-DGateway_Name="OpenMQTTGateway_D1"'
board_build.flash_mode = dout
```

The device sends a JSON output via MQTT and contained in it is the RF value similar to `{"value":1020304}`.

I'll check my wiring again. On the espurna-SRX882 I do have 5VDC between VCC and GND and 3.27VDC on the data pin connected to GPIO 5 on the espurna-wemos. This is the same GPIO I have configured on the OMG-wemos which works with the same type of hardware. I'll try and swap the SRX882s between wemos devices.

davebuk, 2020-07-08:
I was originally trying with my device running PZEMV3 and the SRX882. I have tried again with espurna `rc-switch` and also `OMG rc-switch` deleting the `\.pio\libdeps\db-wemos-rf\rc-switch` folder between builds on a wemos and SRX882 only connected.

Both versions do actually work! Maybe I did have a wiring issue. When I trigger one RF switch I get:

```
[232444] [RF] Received code: 00D7410A
[232445] [RF] Action 0xA4
[232445] [RF] Received message 'C00101C31800D7410A'
[232445] [RF] Trying to match code D7410A
[233024] [RF] Received code: 00D7410A
[233025] [RF] Action 0xA4
[233025] [RF] Received message 'C00101C21800D7410A'
[233025] [RF] Trying to match code D7410A
```
I can set `C00101C31800D7410A` to one of the dummy relays and it switches. My only issue is either the RF switch sends the code twice or the SRX882 sends the data twice. If I set the same code for ON and OFF, the dummy relay just toggles ON/OFF straight away.

I have a test transmission I can run from openHAB that sends an RF code out which this wemos device is receiving. It also gets the data twice.

Are there any filtering options for the received RF codes? Its not too much of a problem for my current applications as the RF door switches I have are two coded, one for open, one for closed.

mcspr, 2020-07-09:
There are a couple of options:
- add frequency configuration to the rfbridge itself, sort of like flood window for relays, to only process non-repeating codes.
- add 'locking' to more places, making relay switch lock itself for some time to avoid switching too fast. right now we have 'flood window' that delays (but not cancels!) the action when doing something with relay too fast, perhaps it could have lock instead of delay (and also b/c existing interlock timer only works with 'sync one' or 'sync none or one' modes)
- provide a slightly different interface to rfbridge through rpn rules. for example, rule like `1 "00D7410A" rfb_match 2 0 relay` would translate to 'when code "D7410A" is detected the very first time in N seconds, toggle the 1st relay'. (N seconds is configurable, 1 at the beginning is what number of repeats we want, 2 is toggle and 0 relay is 1st relay action)

davebuk, 2020-07-09:
Can the 'debouce' code used for event sensors etc be used? Would say a 500ms debounce allow the dummy relay time to ignore any other received codes while processing the first received code?

mcspr, 2020-07-13:
True, it could. Right now I already have the mentioned rpn solution running, with every code time and how much we've seen this code in the last N seconds remembered. `rfb_match` above would just check for additional param of debounce time.

Those are rules though, but it could see it as a 3rd setting for the code defaulting to no debounce time. UI would be weird looking though :/ But, it could also be just a setting without any web presence.

davebuk, 2020-07-16:
I'll give the RPN rules a try.

closed 2020-07-16 by davebuk

mcspr, 2020-07-16:
Let me rebase the patch first though! It is not yet in the tree or as a PR

davebuk, 2020-07-17:
Sorry, I thought it was already part of the current RPN code.