xoseperez/espurna #2297: Shelly i3 support

andreaconfa, 2020-07-06:
Hi, is planned to add support to the new Shelly i3 that is a device with 3 input and 0 output?
Eventually i have here a Shelly i3 if someone wants to make some tests

mcspr, 2020-07-07:
We need real GPIO numbers like here to map things properly:
https://github.com/xoseperez/espurna/blob/009bdf05efb87b4818b21ef7c1326ccb83496737/code/espurna/config/hardware.h#L3615-L3636
Which can be traced on the board itself / read with `gpio <pin-number>` on firmware without anything configured

lociii, 2020-07-19:
[image]
[image]