xoseperez/espurna #2305: Can't connect or Scan Wifi networks when SSID with Special Characters present

rylangrant, 2020-07-19:
**Bug description**
Performing a Wifi Network scan in the WebUI hangs and never completes. In the debug console in the WebUI, performing a wifi.scan hangs and never completes (says it finds a bunch of networks, but never shows the full number found). Through telnet, performing a wifi.scan command does show the full list of networks. In the list, I found a SSID that never shows up when performing scan's in WebUI, and it appears to have some kind of special character at the end of it (See attached picture). This character doesn't show up on my PC or phone when doing a scan. From the name of the SSID, it appears to be an HP printer but this printer isn't mine so I can't unplug to confirm.

I have tried manually specifying the SSID that i want to connect to in the WebUI but it will not connect to it, I'm guessing because the scan function to connect to it fails. 

**Steps to reproduce**
Have a Wifi SSID present that has special characters in it, and perform a Wifi Scan in the Web UI. Web UI hangs during a scan, if manually specify SSID / password the device will never connect to it.

*Steps to reproduce the behavior.*

**Expected behavior**
Scan should not crash when parsing SSID with a special character - omit the character.

*A clear and concise description of what you expected to happen.*
The SSID with the special character should show up in the list as normal and be safely parsed

**Screenshots**
Attached is a screenshot from terminal

**Device information**
---8<-------

[222409] [MAIN] ESPURNA 1.14.1 (62ad7da3)
[222409] [MAIN] xose.perez@gmail.com
[222409] [MAIN] http://tinkerman.cat

[222410] [MAIN] CPU chip ID: 0x6AA23F
[222413] [MAIN] CPU frequency: 80 MHz
[222416] [MAIN] SDK version: 1.5.3(aec24ac9)
[222420] [MAIN] Core version: 2.3.0
[222426] [MAIN] Core revision: 9826c6d
[222427] [MAIN] Build time: 1577761663
[222432] 
[222432] [MAIN] Flash chip ID: 0x144051
[222435] [MAIN] Flash speed: 40000000 Hz
[222438] [MAIN] Flash mode: DOUT
[222441] 
[222442] [MAIN] Flash size (CHIP)   :  1048576 bytes /  256 sectors (   0 to  255)
[222453] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[222458] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[222464] [MAIN] Firmware size       :   512512 bytes /  126 sectors (   1 to  126)
[222471] [MAIN] Max OTA size        :   507904 bytes /  124 sectors ( 127 to  250)
[222478] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[222485] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[222492] 
[222493] [MAIN] EEPROM sectors: 251, 250
[222500] [MAIN] EEPROM current: 251
[222500] 
[222501] [MAIN] EEPROM:  4096 bytes initially |   288 bytes used ( 7%) |  3808 bytes free (92%)
[222511] [MAIN] Heap  : 33936 bytes initially | 20200 bytes used (59%) | 13736 bytes free (40%)
[222517] [MAIN] Stack :  4096 bytes initially |  1872 bytes used (45%) |  2224 bytes free (54%)
[222526] 
[222526] [MAIN] Boot version: 7
[222529] [MAIN] Boot mode: 1
[222534] [MAIN] Last reset reason: Reboot from web interface
[222537] 
[222538] [MAIN] Board: ARILUX_AL_LC06
[222543] [MAIN] Support: ALEXA API BROKER BUTTON DEBUG_SERIAL DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT LED MDNS_SERVER MQTT NTP SCHEDULER TELNET TERMINAL THINGSPEAK WEB 
[222556] [MAIN] OTA: ARDUINO ASYNCTCP WEB 
[222560] [MAIN] WebUI image: LIGHT
[222563] 
[222564] [MAIN] Firmware MD5: f2b66f1801bdd0afeabe6e9a50ff6597
[222573] [MAIN] Power: 3155 mV
[222573] [MAIN] Power saving delay value: 10 ms
[222578] 

---8<-------

[222579] +OK

**Tools used**
Linux Ubuntu - Firefox - 72.01
Windows 10 - Chrome 83.0 / Edge

**Additional context**
It appears the HP direct Wifi broadcasting SSID causes issues. I did try other firmware (Tasmota) and it appears to handle the SSID ok.
[image]
[image]

mcspr, 2020-07-19:
The character isn't the issue, length is. The SSID in question is exactly 32 bytes and underlying 2.3.0 Arduino Core API does not properly handle such long name.

Upgrading it will fix output / scan problem though. Or, disabling the scan functionality, it will just fall back to a simple connection attempt.

rylangrant, 2020-07-19:
How do I upgrade the core? Is there a binary with it or some other process?

Thanks

mcspr, 2020-07-19:
Only full re-build, check out https://github.com/xoseperez/espurna/wiki/PlatformIO
Base env should be esp8266-1m-latest-base

mcspr, 2020-09-10:
Closing via #2333

closed 2020-09-10 by mcspr