xoseperez/espurna #2323: Create a metrics page for prometheus scraping

weikinhuang, 2020-08-15:
**Is your feature request related to a problem? Please describe.**
I would like to be able to have prometheus scrape metrics of my IOT devices such as outlets' power usage.

**Describe the solution you'd like**
Present a page that can be enabled that would output all stats available for prometheus to scrape.

**Describe alternatives you've considered**
Right now I have a node server that aggregates all the api calls that are available for reading power, voltage, total kw since power on, etc and formats them into a prometheus scrape-able format.

**Additional context**
*Add any other context or screenshots about the feature request here.*

mcspr, 2020-08-15:
Can you share the adapter layer? What format is required, exactly?

Also note that after https://github.com/xoseperez/espurna/commit/f3c185cc7367ed6cf3cef0887a7d5dc01acea197 you can call `/magnitudes?apikey=...` with tuples of `["magnitude-name", last-reading, last-report]`. I am not attached to the scheme used there

weikinhuang, 2020-08-16:
The output should look like this, the proxy server is just a simple node app that makes use of the prometheus lib, and a few http requests

```
# HELP espurna_relay0 Relay 0 is on
# TYPE espurna_relay0 gauge
espurna_relay0 1

# HELP espurna_current The current (A)
# TYPE espurna_current gauge
espurna_current 2.012

# HELP espurna_voltage The voltage (V)
# TYPE espurna_voltage gauge
espurna_voltage 120

# HELP espurna_power Real power usage (W)
# TYPE espurna_power gauge
espurna_power 225

# HELP espurna_reactive Volt Amp reactive power
# TYPE espurna_reactive gauge
espurna_reactive 92

# HELP espurna_apparent The magnitude of complex power
# TYPE espurna_apparent gauge
espurna_apparent 241

# HELP espurna_factor The ratio of active power to apparent power
# TYPE espurna_factor gauge
espurna_factor 92

# HELP espurna_energy Total energy usage since boot (kWh)
# TYPE espurna_energy counter
espurna_energy 577.722
```

The documentation is located here, https://prometheus.io/docs/instrumenting/writing_exporters/ but the general idea is:
```
# Description
# Data type
name{label=foo} VALUE
```

weikinhuang, 2020-08-18:
The `HELP` and `TYPE` comments are optional

mcspr, 2020-08-18:
So the plan is to add:
- `<host>:80/metrics` endpoint
- endpoint handler that gathers relays + sensors (+ other relevant info?) as `<name> <space> <value>`
backbone sort of like the existing Homeassistant discovery (see HASS in WebUI and generate cfg as an example), but with up-to-date data

Do we need `espurna_...` prefix, or have it configurable / at all?
Suffixes are weird, too, if prefixes are used - we need some other name for `MAGNITUDE_COUNT` b/c we will use the built-in name `"count"`

weikinhuang, 2020-08-18:
no prefix needed, that's totally optional, it doesn't need to be configurable, as prometheus usually adds additional labels to the data

closed 2020-08-30 by mcspr

weikinhuang, 2020-08-30:
Thanks @mcspr!

mcspr, 2020-08-30:
Mini update in dev, latest commit adds additional check for initial values on device boot so it does not report them as nan / 0.0 and wait until they switch to numeric value.
Also note that values are based on `apiRealTime` setting:
- when `0` it returns latest reading
- when `1` - latest 'report' (i.e. filtered value, based on the `snsReport` counter)

PR #2332 (2020-08-30): web: prometheus metrics support
resolve #2323 

Looks like:
```
relay0 0
relay1 0
relay2 0
relay3 0
temperature0 10.000
temperature1 9.000
humidity 35.000
pressure 997.000

```

I'll need to adjust sensor value printing, decimal places are hard-coded to 3 and need to be handled by the sensors module (somehow), as those methods aren't public right now.
<s>HTTP auth won't work, as it seems. I can't find any config option to accept user:pass pair for the metrics target.</s>

```yaml
scrape_configs:
  - job_name: 'espurna'
    metrics_path: '/api/metrics'
    params:
      apikey: ['apikeyapikey'] [... cut]