xoseperez/espurna #2324: SonOff RF Bridge reloads whit mqtt <root>/rfout/set message

jvilafe, 2020-08-16:
Hi,

we have one SonOff RF bridge 433Mhz with ESPurna 1.14.2-dev firmware. No direct-hack.
We want  mqtt-to-RF and RF-to-mqtt operation.

I can see the mqtt <root>/rfin/ message when I activate any RF devices, this feature works fine but...
when I send mqtt <root>/rfout/set message, it doesn't matter the payload of the message, the RF board reloads. I can't see any message in the GUI debug window because of the reload process.

Can anybody helps?

mcspr, 2020-08-16:
> ESPurna 1.14.2-dev firmware. 

Have you tried updating to the current version from the dev branch? Can't reproduce this issue.

jvilafe, 2020-08-18:
Hi mcspr,

I update to 1.15.0-dev. Same issue, when I publish anything with topic <root>/rfout/set, de board reloads.
[image]

mcspr, 2020-08-18:
Does it work after removing these 3 lines?
https://github.com/xoseperez/espurna/blob/0f11b662a1221bfd328706775796e4495563367b/code/espurna/rfbridge.cpp#L417-L419

jvilafe, 2020-08-18:
WOW!!

yes, it didn't reload without these lines!

thank you!

jvilafe, 2020-08-18:
Hi again,

my board stops working!! It doens't responds to gui...I unplug power and plug it again...the board only start logo led...no GUI, no ping (no ARP in table for its IP) Wifi led off.
I tried to reflash but I can't connect...It seams that UART isn't responding...I'm using NodeMCU flasher, Tasmotizer, Platformio with VSCode...all methods with the same behavior...connecting to board and No responses.