xoseperez/espurna #2325: Flash Shelly devices without opening

yaourdt, 2020-08-17:
**Is your feature request related to a problem? Please describe.**
I've added espurna to my intermediate firmware for OTA flashing Shelly devices with open source firmware. You can find [mg2x here](https://github.com/yaourdt/mgos-to-tasmota) and have it flash espurna directly from Shelly stock firmware.

**Describe the solution you'd like**
If you think this is useful to your users, we could add it to the docs.

**Describe alternatives you've considered**
If you think it's useless, don't add it ;)

**Additional context**
I'd be happy to contribute to the doc if you decide to add it.

mcspr, 2020-08-18:
Sure. Something similar to the https://github.com/xoseperez/espurna/wiki/OTA-TUYA?

I noticed that you use 1mb base image, does this work when OTA'ing to the shelly25 .bin which uses 2mb settings? Arduino OTA Updater is very particular about size differences, not really sure it would accept up-sizing based on sdk size setting even if the flash chip size allows it

yaourdt, 2020-08-18:
> Sure. Something similar to the https://github.com/xoseperez/espurna/wiki/OTA-TUYA?

Yes, that's what I thought. 

You are right about the 1 MB image, moving from there to a 2 MB image may be difficult via OTA. The reason I chose the 1MB base image is that I could not find a `espurna-1.14.1-espurna-base-2MB.bin` image, and I was not certain if `espurna-1.14.1-espurna-core-2MB.bin` was a good choice, as TUYA CONVERT also uses the base image.

mcspr, 2020-08-18:
I might be wrong though (but I guess I could test with d1 mini or something similar and try to upgrade 1mb base to 4mb base, if I am missing some other code checking this condition):
https://github.com/esp8266/Arduino/blob/1ff927d04caa28e1294e94ec10edf758bec6edf1/cores/esp8266/Updater.cpp#L469
Technically, usable size is the same for ours 1mb and 2mb is the same, as Xose originally used 1mb + 1mb split. Last 1mb are reserved for the FS partition, but it does not do anything atm.

Current `dev` branch is relatively stable for the purposes of the base image, see:
https://github.com/mcspr/espurna-nightly-builder/releases
-base was renamed into -core-webui, + I have added 2mb variant. -core does not have WebUI, only OTA available there is via [espota.py](https://github.com/esp8266/Arduino/blob/master/tools/espota.py). Most (is not all) reported Tuya devices were mostly esp8285 with 1mb chips, so we were safe so far.

yaourdt, 2021-01-18:
@mcspr I'm very sorry for not answering in such a long time, I completely forgot about this issue, my apologies!

I just tried moving from `espurna-1.14.1-espurna-base-1MB.bin` to `espurna-1.15.0-dev.nightly20210118+git8e659c94-espurna-core-webui-2MB.bin`, and it worked, except for the fact that the device was reset to factory settings (I had configured an admin PW and WiFi credentials). Which was to be expected, given the fact that the system parameters area address changes in the process.

So, moving to larger images should be relatively safe (it is usually, as long as the system configuration parameters are considered, which seems to be the case here).

I think the best option is:
 1. Move from Mongoose to Espurna base 1MB
 2. Configure admin PW
 3. Switch to desired version of Espurna, depending on device and flash size
 4. Configure admin PW again, along with all the rest

In my opinion, it makes no sense to move to a preconfigured version of Espurna directly, as there are too many options to choose from.

mcspr, 2021-01-19:
Seems like a waste doing this password change twice though, I'll change the -base configuration to disable the webui must-change-default-password build flag
(fwiw, password change could be skipped if user chooses to use /upload endpoint directly)

I have also added a direct link to the repo in the wiki's sidebar some time ago :)

yaourdt, 2021-01-19:
Nice, I did not know. In this case, yes, there is no need to set it twice. I'll add it to my firmwares README.md

I saw that, it's an honor 😌

closed 2021-01-21 by yaourdt