xoseperez/espurna #2328: Add a delay parameter in the relay control API endpoints

bderancourt, 2020-08-19:
Hi !

Let's me explain the context:
I have a 3D printer controlled by a [raspberry pi](https://octoprint.org). It's common on the 3D printing world.
I put a sonoff basic on my 3D printer to be able to start/stop my 3D printer remotely. Start works great but shutdown is a little more complicated. To be perfect, I should switch off the power when the raspberry pi OS has shut down properly. So I need to send a 'turn off' request to the sonoff with a little delay (30s).

I searched a solution with espurna and I didn't find a way to do it. There is the schedule feature but that's not exactly what I need.

So I suggest to add a delay parameter on the PUT /api/relay/x endpoint.
Ideally the endpoint response will return the delay parameter but that's not mandatory.

Ex:
`$ curl -X PUT -H "Accept: application/json" http://192.168.1.108/api/relay/0 --data "apikey=C62ED7BE7593B658&value=2"
{ "relay0": 0, "delay": 30}`

And ideally, if a delayed call has been made, the GET endpoint should return information about it:
`$ curl -H "Accept: application/json" http://192.168.1.108/api/relay/0?apikey=C62ED7BE7593B658
{ "relay0": 1, "delay": 23, "action": 2} // action or value or newValue, to be decided.`
This part is optional.

Thank you !

mcspr, 2020-08-19:
This is something that was added by Xose originally, see `/api/pulse/<relay_id>` (note that we don't suffix the relay, but have a whole different endpoint). Sending time in in seconds will flip the specified relay ID status and wait, then switch back to the previous state after the timer expires. Sub-second delays should use decimal point e.g. 0.5 for 500 ms

As a sidenote, idk if having JSON endpoints for a single entries is justified, since we already know the entity by the path and just waste time on parsing json :/ V*eee*ry recent commit adds `/api/relay` that only accepts GET at this time, but perhaps it could work with PUT and even allow to modify multiple relays in bulk.

bderancourt, 2020-08-19:
Thank you for your answer.

I tried the pulse API but it's not suitable for my needs as it flip the relay immediatly the first time it received the call. It would be adapted if the first flip was absent. I just need to delay the action.

For your remark about JSON endpoints, the JSON part is the answer. I correct the typo.

bderancourt, 2020-08-23:
Thanks for the tip.

So I flash the 1.14.1 release with this custom.h file:
```
#if defined(MY_SONOFF_ENDER3)    // This section will be used when src_build_flags contains
                                  //  -DMY_SONOFF_BUILD01 in 'platformio_override.ini'

    // Info
    #define MANUFACTURER        "ITEAD"
    #define DEVICE              "SONOFF_BASIC"

    // Buttons
    #define BUTTON1_PIN         0
    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON1_RELAY       1
    #define BUTTON2_PIN         14
    #define BUTTON2_MODE        BUTTON_SWITCH | BUTTON_SET_PULLUP | BUTTON_DEFAULT_HIGH
    #define BUTTON2_RELAY       1

    // Relays
    #define RELAY1_PIN          12
    #define RELAY1_TYPE         RELAY_TYPE_NORMAL

    // LEDs
    #define LED1_PIN            13
    #define LED1_PIN_INVERSE    1

    // Disable these
    #define DEBUG_SERIAL_SUPPORT       0
    #define ALEXA_SUPPORT              0
    #define DOMOTICZ_SUPPORT           0
    #define HOMEASSISTANT_SUPPORT      0
    #define THINGSPEAK_SUPPORT         0
    #define SCHEDULER_SUPPORT          0
    
    // Custom settings
    #define RELAY1_DELAY_OFF           30000
    #define ADMIN_PASS                 "xxx"
    #define WEB_FORCE_PASS_CHANGE      0
    #define WIFI_SCAN_NETWORKS         0
    #define WIFI1_SSID                 "xxx"
    #define WIFI1_PASS                 "xxx"
    #define NTP_SERVER                 "xxx"

#endif
```

It worked for one day then it starts to malfunction: The sonoff was plugged to 220V but the button did not turn the relay on and seems to be inactive. Same on the UI. I needed to wait for a few minutes to be able to switch the relay again. Maybe this delay function has not been tested much ?
I flashed a version without the RELAY1_DELAY_OFF setting and all started to work again.

Despite this workaround that seems to have unwanted side effects at this time, it would be better and simpler to have this parameter directly in the API. Please :innocent:

mcspr, 2020-08-27:
Delay **is** relay's property, thus button press will trigger the delay as well (both directions). You could check what debug log says btw; relay module is very chatty about it's operation, when & how it schedules it's actions.

It would be simpler from the API, true. Given the amount of timers that we have it seems very fragile though >_>, I tried to hack something that might fit the use-case (but not tested, yet):
```diff
diff --git a/code/espurna/relay.cpp b/code/espurna/relay.cpp
index 7dfadf16..bd10d2a5 100644
--- a/code/espurna/relay.cpp
+++ b/code/espurna/relay.cpp
@@ -1065,6 +1065,32 @@ void relaySetupAPI() {
             }
         });

+        sprintf_P(path, PSTR("pulse_delay/%u"), id);
+        apiRegister({
+            path, Api::Type::Basic, id,
+            [](const Api& api, ApiBuffer& buffer) {
+                snprintf(buffer.data, buffer.size, PSTR("on:%u,off:%u"),
+                    _relays[api.arg].delay_on,
+                    _relays[api.arg].delay_off
+                );
+            },
+            [](const Api& api, ApiBuffer& buffer) {
+                unsigned long ms = 1000ul * atof(buffer.data);
+                if (!ms) return;
+
+                auto& target = relayStatus(api.arg)
+                    ? _relays[api.arg].delay_off
+                    : _relays[api.arg].delay_on;
+
+                // relayToggle will set change_delay=target
+                // hijack it while it is doing it and replace with the existing value after that
+                auto tmp = target;
+                target = ms;
+                relayToggle(api.arg);
+                target = tmp;
+            }
+        });
+
         sprintf_P(path, PSTR(MQTT_TOPIC_PULSE "/%u"), id);
         apiRegister({
             path, Api::Type::Basic, id,
```
Sets up a new `pulse_delay/#`, which works the same as `pulse/#`, but delays the action instead of setting up a timer to toggle back (and it is always a toggle, as the opposite option does not really make sense here; we would end up doing what runtime settings already do)

Which API approach is better is another question...

closed 2024-08-23 by mcspr