xoseperez/espurna #2329: RPN rules Enable not possible

HilverinkJ, 2020-08-19:
After flashing itead-sonoff-th I found out that i have to enable RPN rules. In order to activate this option I sed #define RPN_RULES_SUPPORT    to    1.
After flashing with this new setting, the website failed to open.
The serial monitor is running.
Wath to do?

mcspr, 2020-08-19:
Template is there for a reason, *what* version do you use and what build parameters?

If this is some kind of weird WS / JS error, web browser's js console will surely have an answer.
If `heap` command shows values below 10k, this most likely is OOM silently dropping the request. Or, if not, tcpdump / wireshark will show the actual traffic that happens before it stops (as we did have some issues with json unicode serializer dumping garbage on the wsocket and browser helpfully shutting the whole connection down)

HilverinkJ, 2020-08-19:
After removing all the ESPurna code and cloning the latest version to my system, I made only these changes:
hardware.h
#elif defined(ITEAD_SONOFF_TH)
removing DHT support
    //#ifndef DHT_SUPPORT
    //#define DHT_SUPPORT         1
    //#endif
    //#define DHT_PIN             14
    //#define DHT_TYPE            DHT_CHIP_SI7021
[image]
This is Without RPN support............

Next changing
general.h
#define RPN_RULES_SUPPORT           1               // Enable RPN Rules (8.6Kb) 
setting to enable.

[image]
After upload end monitor the firmware it looks like this.
> ===================================================================================================== [SUCCESS] Took 80.32 seconds =====================================================================================================
--- Available filters and text transformations: colorize, debug, default, direct, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters
--- Miniterm on COM6  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
[059749] [RELAY] Setting relay mask: 0b0
[060006] [MAIN] System OK
[061041] [WIFI] --> BSSID: 34:31:C4:BB:E6:5F CH:  1 RSSI: -73 SEC: WPA2 SSID: AllFindOn
[061041] [WIFI]     BSSID: 36:31:C4:BB:E6:5F CH:  1 RSSI: -73 SEC: WPA2 SSID: AllFindOn guest access      
[061046] [WIFI]     BSSID: 78:96:82:43:36:B1 CH:  1 RSSI: -84 SEC: WPA2 SSID: H369A4336B1
[061054] [WIFI]     BSSID: 44:94:FC:44:21:38 CH:  1 RSSI: -82 SEC: WPA2 SSID: ZiggoDCA54_EXT
[061062] [WIFI]     BSSID: 10:7B:44:58:C4:48 CH:  1 RSSI: -90 SEC: WPA2 SSID: Ons Thuus
[061070] [WIFI] --> BSSID: E8:DF:70:A4:66:8E CH:  4 RSSI: -40 SEC: WPA2 SSID: AllFindOn
[061077] [WIFI]     BSSID: 6A:DF:70:A4:66:8F CH:  4 RSSI: -40 SEC: WPA2 SSID: AllFindOn guest access      
[061086] [WIFI]     BSSID: 06:18:D6:4F:B9:36 CH:  1 RSSI: -91 SEC: WPA2 SSID:
[061093] [WIFI]     BSSID: 04:18:D6:4F:B9:36 CH:  1 RSSI: -90 SEC: WPA2 SSID: Stichting Peuterwerk - PSZ  
[061102] [WIFI]     BSSID: D8:5D:E2:02:3C:1D CH: 11 RSSI: -69 SEC: AUTO SSID: HP-Print-1D-LaserJet Pro MFP
[061111] [WIFI]     BSSID: 20:E8:82:BF:55:80 CH: 11 RSSI: -60 SEC: WPA2 SSID: H369ABF5580
[061119] [WIFI]     BSSID: FC:C8:97:8A:32:91 CH: 11 RSSI: -86 SEC: AUTO SSID: H220N8A3291
[061127] [WIFI]     BSSID: 62:C8:97:8A:32:92 CH: 11 RSSI: -87 SEC: OPEN SSID: KPN Fon
[061134] [WIFI]     BSSID: 92:5C:14:C7:79:EB CH: 11 RSSI: -90 SEC: OPEN SSID: Ziggo
[061142] [WIFI]     BSSID: 90:5C:44:C7:79:EB CH: 11 RSSI: -92 SEC: WPA2 SSID: nissink
[061149] [WIFI]     BSSID: D4:5D:64:99:9B:78 CH: 11 RSSI: -89 SEC: WPA2 SSID: Asus-HighSpeed-WiFi
[061169] [WIFI] Connecting to BSSID: E8:DF:70:A4:66:8E CH: 04, RSSI: -40, SEC: WPA2, SSID: AllFindOn      
[062064] [WIFI] Captive portal disabled
[062064] [WIFI] ------------------------------------- MODE STA
[062064] [WIFI] SSID  AllFindOn
[062065] [WIFI] IP    192.168.178.82
[062068] [WIFI] MAC   A4:CF:12:B5:AB:71
[062071] [WIFI] GW    192.168.178.1
[062074] [WIFI] DNS   192.168.178.1
[062077] [WIFI] MASK  255.255.255.0
[062081] [WIFI] HOST  http://ESPURNA-B5AB71.local
[062085] [WIFI] BSSID E8:DF:70:A4:66:8E
[062088] [WIFI] CH    4
[062090] [WIFI] RSSI  -44
[062093] [WIFI] ----------------------------------------------
[062099] [MDNS] OK
[063101] [MQTT] MQTT brokers found: 0
[067142] [NTP] UTC Time  : 2020-08-19 18:34:29
[067142] [NTP] Local Time: 2020-08-19 20:34:29
[123168] [WEBSERVER] Request: GET /index.html
[300004] [MAIN] Uptime: 00d 00h 05m 00s
[300004] [MAIN] Heap  : 26352 bytes initially | 21704 bytes used (82%) |  4648 bytes free (17%)
[300005] [MAIN] Power: 4094 mV
[300007] [MAIN] Time: 2020-08-19 20:38:21
[600009] [MAIN] Uptime: 00d 00h 10m 00s
[600010] [MAIN] Heap  : 26352 bytes initially | 21704 bytes used (82%) |  4648 bytes free (17%)
[600011] [MAIN] Power: 4094 mV
[600013] [MAIN] Time: 2020-08-19 20:43:21

[image]

mcspr, 2020-08-19:
Can you try it after `-t build-lwip`, as described here? (won’t work with Windows though, it needs `make`)
https://github.com/xoseperez/espurna/issues/1610#issuecomment-493974405

mcspr, 2020-08-23:
Can you check MTU setting on the computer in question?

closed 2020-08-23 by mcspr