xoseperez/espurna #2330: RFB_DIRECT ignores RF_SEND_TIMES

franklysaputra, 2020-08-19:
Hi!

I have modified my Sonoff RF Bridge with DIRECT Hack. The problem is RF_SEND_TIMES is completely ignored and always defaulted to 1.

**custom.h:**

```
#define ALEXA_SUPPORT           0
#define DOMOTICZ_SUPPORT        0
#define HOMEASSISTANT_SUPPORT   0
#define THINGSPEAK_SUPPORT      0
#define RPN_RULES_SUPPORT       0
#define RFB_DIRECT              1
#define RF_SEND_TIMES           4
#define DUMMY_RELAY_COUNT       10
#define RF_SEND_DELAY           200
#define RF_RECEIVE_DELAY        500
```

**Terminal output (telnet):**
```
[223753] [MQTT] Received RFBridge/rfout/set => C00101591800ECDB88
[223758] [RF] Trying to match code ECDB88
[223763] [RF] Enqueuing MESSAGE '??"@?2?' 1 time(s)
```

**Device Info:**
```
---8<-------

[023560] [MAIN] ESPURNA 1.15.0-dev (0f11b662)
[023560] [MAIN] xose.perez@gmail.com
[023561] [MAIN] http://tinkerman.cat

[023564] [MAIN] CPU chip ID: 0x4349DF
[023568] [MAIN] CPU frequency: 80 MHz
[023571] [MAIN] SDK version: 1.5.3(aec24ac9)
[023575] [MAIN] Core version: 2.3.0
[023581] [MAIN] Core revision: 9826c6d
[023581] [MAIN] Build time: 1597869500
[023585] 
[023586] [MAIN] Flash chip ID: 0x144051
[023591] [MAIN] Flash speed: 40000000 Hz
[023595] [MAIN] Flash mode: DOUT
[023596] 
[023597] [MAIN] Flash size (CHIP)   :  1048576 bytes /  256 sectors (   0 to  255)
[023607] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[023611] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[023620] [MAIN] Firmware size       :   491344 bytes /  120 sectors (   1 to  120)
[023625] [MAIN] Max OTA size        :   532480 bytes /  130 sectors ( 121 to  250)
[023632] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[023640] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[023648] 
[023648] [MAIN] EEPROM sectors: 251, 250
[023654] [MAIN] EEPROM current: 250
[023655] 
[023656] [MAIN] EEPROM:  4096 bytes initially |   414 bytes used (10%) |  3682 bytes free (89%)
[023666] [MAIN] Heap  : 30488 bytes initially | 16920 bytes used (55%) | 13568 bytes free (44%)
[023673] [MAIN] Stack :  4096 bytes initially |  1796 bytes used (43%) |  2300 bytes free (56%)
[023680] 
[023681] [MAIN] Boot version: 31
[023684] [MAIN] Boot mode: 1
[023689] [MAIN] Last reset reason: Reboot from web interface
[023692] 
[023693] [MAIN] Board: ITEAD_SONOFF_RFBRIDGE
[023699] [MAIN] Support: API BROKER BUTTON DEBUG_SERIAL DEBUG_TELNET DEBUG_WEB LED MDNS_SERVER MQTT NTP RF SCHEDULER TELNET TERMINAL WEB 
[023709] [MAIN] OTA: ARDUINO ASYNCTCP WEB 
[023712] [MAIN] WebUI image: RFBRIDGE
[023715] 
[023948] [MAIN] Firmware MD5: c3c1add0382edc8cd5f0b5543684ab97
[023954] [MAIN] Power: 3112 mV
[023954] [MAIN] Power saving delay value: 10 ms
[023954] 

---8<-------
```

Any help would be greatly appreciated! Thank you

mcspr, 2020-08-19:
It is not fully ignored, just has a different connotation:
https://github.com/xoseperez/espurna/blob/0f11b662a1221bfd328706775796e4495563367b/code/espurna/rfbridge.cpp#L412
https://github.com/xoseperez/espurna/blob/0f11b662a1221bfd328706775796e4495563367b/code/espurna/rfbridge.cpp#L784

Perhaps, these are wrong, and repeats should actually mean queued repeats as with the sonoff rfbridge software (idk what transmit value is there though)

franklysaputra, 2020-08-20:
@mcspr thanks for your explanation. So is it possible to set the N times in payload value to override the default ```times = 1``` in rfbridge.cpp?

mcspr, 2020-08-20:
After erasing `#if RFB_DIRECT ... #endif` block, sending `<payload:hex-string>,<times:number>` will override the default 1.

mcspr, 2020-08-20:
Also, re:
> **Terminal output (telnet):**
> ```
> [223753] [MQTT] Received RFBridge/rfout/set => C00101591800ECDB88
> [223758] [RF] Trying to match code ECDB88
> [223763] [RF] Enqueuing MESSAGE '??"@?2?' 1 time(s)
> ```

See https://github.com/xoseperez/espurna/issues/2324#issuecomment-675366175, there is (and was for a while) a bug with the  debug print. I'd suggest to also remove those lines in addition to the `#if ... #endif` from above.

franklysaputra, 2020-08-26:
@mcspr thank you so much 😁😁

mcspr, 2020-08-27:
Also just merged #2335 with these issues fixed. Repeats setting is always override-able, both from settings and MQTT.

Only difference is codes like this:
```
C00101591800ECDB88
```
Become like this:
```
C001015918ECDB88
```
0x18 byte is bit length -> 24 bits -> 3 bytes of payload to the left.

In case you have some saved codes, update *should* convert existing ones. But, before OTA'ing / flashing it, check out what `get cfg` terminal output is. If it is 4, migrate will do it's thing. If empty, manually set it to 4 via `set cfg 4` and then flash.

franklysaputra, 2020-08-27:
I clean flashed my RF Bridge and I can confirm that changing repeats in web GUI is working only if you reboot the device. But I can't get the repeats overriden by MQTT with this e.g. payload:
```C001015918ECDB88,10```

mcspr, 2020-08-27:
It should work now, check out the latest commit.

closed 2020-09-04 by franklysaputra