xoseperez/espurna #2331: Sonoff Touch se encienden aleatoriamente

tonymerce, 2020-08-21:
Hola, actualmente tengo en mi casa con espurna:
23 interruptores sonoff touch, T1 2c, T1 3c
5 sonoff mini
2 sonoff S20

Tengo 9 de ellos que se encienden aleatoriamente, el problema no es de espurna ya que alguno lo hacían con el software original. Los que fallan mayoritariamente son los touch aunque he detectado algún fallo mucho mas esporádico con algún T1 2c.

En los dispositivos que no son tactiles, los mini y los S20, no hay ningún problema.

Todos los dispositivos están conectados al servidor mosquitto (windows 10) y de ahi al domoticz (windows 10), ademas tengo instalado homebridge (Windows 10) para que todos funcionen en homekit.

He probado todas las versiones de espurna, el borrado mosquitto.db y he quitado retain en todos los dispositivos. Tengo claro que es un problema hardware pero quería saber si a alguien mas le pasa o le ha pasado y como lo ha solucionado. Tengo 9 interruptores que no puedo usar para nada ya que tienen este fallo. He encontrado información de estos encendidos "fantasma" en los foros de TASMOTA pero no en los foros de ESPURNA.

Perdonar que escriba en Español, no quiero usar el google para traducir y no quede claro lo que quiero decir.
Gracias por anticipado.

tonymerce, 2020-08-24:
1.- No tengo experiencia con Tasmota, solo he probado versiones de Espurna desde la 1.13 hasta la 1.15.
2.- Todos los dispositivos esta en la misma Wifi de 2.4 Ghz
3.- No tengo ninguna Subred en la casa.

Gracias por contestar.

mcspr, 2020-08-31:
Can you retry with the current nightlies / dev builds?

oscarsan1, 2020-09-17:
Yo tambien he tenido algun problema como el tuyo. Los botones del touch son demasiado sensibles y a veces disparan solos. Con algun otro he tenido el problema inverso, tienes que pulsar demasiado fuerte para que te detecte. Prueba a quitar la tapa, deberian dejar de fallar, asi lo puedes comprobar. Puedes mirar si se ha deformado el plastico de la carcasa y el protector queda demasiado cerca del tactil y eso hace que se active. 
Pero no parece un problema de firmware, al menos en mi caso no lo era.

tonymerce, 2020-09-18:
Gracias por contestar. Tengo claro que no es del Firm de Espurna pero quería saber si le pasaba a alguien mas. Al final he tenido que quitar los 9 que fallaban y poner nuevos, no se que hacer con ellos :( . Lo de quitarles la tapa ya lo intente y si efectivamente con la tapa quitada no falla, pero no pueden estar con la tapa quitada. Algún otro truco?

pcornill, 2020-09-28:
I have the same issue with one of my Sonoff T3

oscarsan1, 2020-09-28:
Si son muy sensibles puedes probar a poner cinta aislante encima del sensor o en la trasera de la tapa para aumentar la distancia y reducir la sensibilitat. Yo lo hice con cinta de aluminio para el caso contrario de poca sensibilida y en algun caso mejoró.

closed 2020-12-19 by stale[bot]

DMANIAC, 2022-02-21:
I have same issue. In my case, it is with espurna too but it just turns on or off randomly, it does the same with or without the cover.
I've check and it not loses connection at any time.
Some help please.

mcspr, 2022-02-22:
One possible issue is that T1 and later models include RF receiver, that will signal button 'press' after receiving certain RF code. Depending on the variant, either you just hold the button for ~5sec to remove the RF code association. Or, older devices have a reset button under the cover that needs to be held instead. You may want to look at the manual that comes with the device.

pcornill, 2022-02-22:
@mcspr, 5 sec is to pair the RF remote, keep pressing until the second bip (8-10 sec) then produce the RF code to unpair.
If you have an RF bridge, you can even scan for RF codes.
I had the issue and just replace the device. I think it was the hardware that was malfunctioning, not the firmware.

diegoscorpions, 2022-03-06:
El problema es de hardware. Yo lo he "solucionado"  quitando el oscilador de cristal. De encenderse aleatoriamente a todas horas, he pasado a que solo ocurra de vez en cuando...

DMANIAC, 2022-03-06:
I read something about it can be produced by noise or interferences and can be solved by welding some components, but I couldn't understand it clearly.
If someone could have any idea or site for better understanding please?

DMANIAC, 2022-03-06:
> El problema es de hardware. Yo lo he "solucionado" quitando el oscilador de cristal. De encenderse aleatoriamente a todas horas, he pasado a que solo ocurra de vez en cuando...

Es mejor que nada, pero no es una solución. Gracias igualmente.

tonymerce, 2022-03-07:
Yo leí por algún foro pero de Tasmota (Que también pasa con tasmota) que se podía solucionar cambiando el valor de una resistencia, esa resistencia (smd) controlaba la sensibilidad de la pulsación ¿Me gustaría saber donde esta ese oscilador de cristal para hacer alguna prueba?