xoseperez/espurna #2338: Receiving 4 rf codes after rf button press

nedich, 2020-08-30:
Hey guys, 

My setup is Espurna 1.15.0-dev on Sonoff RF Bridge.

On each press of an RF button are received 4 almost identical packets one after another and are then transmitted via MQTT. 

I'm trying to filter out these multiple rf codes so I receive only one in the Nodered. 

As I understand it is some kind of common practice is to send codes 4 times so I guess there is some common practice to filter it but I can't figure out how.

mcspr, 2020-08-30:
Do you mean 4 codes for a single button press? I guess nodered could count things instead of triggering immediately?
I think this was also raised at https://github.com/xoseperez/espurna/issues/2296#issuecomment-656352918 - we don't track repeats in general, but we do that when using [RPN rules](https://github.com/xoseperez/espurna/wiki/RPN-Rules), #2302 
(although, as an example, openmqttgateway mentioned in the other issue actually does via global throttle based on the payload contents. default to 3s at compile time for *every* entry)

Just to note, unique values are protocol and data at the end of the string:
https://github.com/xoseperez/espurna/blob/a409dedefefeeae970abae70b8ecaaefc05bde6a/code/espurna/rfbridge.cpp#L387-L392
(byte [1] is protocol, bytes [5:] is data)

nedich, 2020-08-30:
@mcspr, thanks for the prompt reponse, yes 4 codes per single press. 

The EFM8BB1 (SRFB Stock and Portisch) was doing that filtering implicitly so I thought that Espurna (Pilight?Rcswitch?) may be does that also only I cant find it :) 

Anyway, I'll try those RPN rules you mentioned, i'm new in Espurna so I havent stumbled upon them yet.

Thanks

closed 2020-08-30 by nedich