xoseperez/espurna #2343: Unable to build in platformio

reefwarrior, 2020-09-04:
Hi there,

I have not been able to get the .bin file to be built.

I get an error -> *** [.pio\build\nodemcu-lolin\firmware.bin] TypeError 'ex[ected str, bytes oro os.Pathlike object, not Dir' trying to evaluate '$__get_flash_size(__env__)}'

I am using Windows10 - if that is of any help?

All help appreciated!
Tinus

mcspr, 2020-09-04:
what version of espurna? if git, what `git rev-parse HEAD` shows?
what `pio --version`?

reefwarrior, 2020-09-04:
Hi;

git rev-parse HEAD -> 3d1b82ea258100f462174bf600a4aba73740224d

Don't know which espurna version - I did a git clone. and the pio --version it says it not an internal or external command.

Sorry but I have extremely limited experience with git/etc.

mcspr, 2020-09-04:
np, that is exactly what I wanted to know about our's side of things

for `pio` command, I think it should be in one of these places:
- `%USERPROFILE%\.platformio\penv\Scripts\pio.exe'
- `C:\.platformio\penv\Scripts\pio.exe`

what do
- `pio.exe --version`
- `pio.exe system info`

show?

reefwarrior, 2020-09-04:
awesome!

PlatformIO, version 5.0.0
--------------------------  -------------------------------------------------------------
PlatformIO Core             5.0.0
Python                      3.7.7-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\Tinus Blaauw\.platformio
PlatformIO Core Executable  C:\Users\Tinus Blaauw\.platformio\penv\Scripts\platformio.exe
Python Executable           c:\users\tinus blaauw\.platformio\penv\scripts\python.exe
Global Libraries            0
Development Platforms       2
Tools & Toolchains          10
--------------------------  -------------------------------------------------------------

reefwarrior, 2020-09-04:
[platformio_override.zip](https://github.com/xoseperez/espurna/files/5176193/platformio_override.zip)
[custom.zip](https://github.com/xoseperez/espurna/files/5176196/custom.zip)

Included two files I have modified, if that should be of any help perhaps?

mcspr, 2020-09-04:
can you try again?

`system info` given out the reason - spaces in absolute filepaths... I tried renaming the working repo dir, which triggered the issue you described. should work now

closed 2020-09-04 by mcspr

reefwarrior, 2020-09-04:
PermissionError: [WinError 5] Access is denied: 'C:\\Users\\Tinus Blaauw\\.platformio\\.cache\\tmp\\pkg-installing-i2fuchvm\\.git\\objects\\pack\\pack-1e52effe1f9a5937ec9554213076ce1b1e541f6a.idx':
  File "C:\users\tinus blaauw\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 169:

commit e1b643fd (2020-09-04): pio: update ldpath as soon as possible

trying to fix #2343
*** [.pio\build\$PIOENV\firmware.bin] TypeError
`expected str, bytes or os.PathLike object, not Dir' trying to evaluate `${__get_flash_size(__env__)}'

as it turns out, PIO platform internals (sometimes) did not use the correct ldscript path