xoseperez/espurna #2344: can you try again?

reefwarrior, 2020-09-05:
can you try again?

`system info` given out the reason - spaces in absolute filepaths... I tried renaming the working repo dir, which triggered the issue you described. should work now

_Originally posted by @mcspr in https://github.com/xoseperez/espurna/issues/2343#issuecomment-687355748_

reefwarrior, 2020-09-05:
Sorry, as this is a duplicate - the original 'ticket' was closed, so I don't know if you were able to get it @mcspr  :)

mcspr, 2020-09-05:
#2343 was closed via https://github.com/xoseperez/espurna/commit/e1b643fd0021692cf1a247a6e527068d08cdaac2
have you 'pull'ed from the repo again to sync it?

> PermissionError: [WinError 5] Access is denied: 'C:\Users\Tinus Blaauw\.platformio\.cache\tmp\pkg-installing-i2fuchvm\.git\objects\pack\pack-1e52effe1f9a5937ec9554213076ce1b1e541f6a.idx':
File "C:\users\tinus blaauw.platformio\penv\lib\site-packages\platformio\builder\main.py", line 169:

does not tell me anything useful :/
if you happen to run vscode, try again with `pio.exe` directly? (`pio.exe run -e $PIO_ENVIRONMENT_NAME`)

reefwarrior, 2020-09-05:
Tried. Get the same problem about PermissionError :(

reefwarrior, 2020-09-05:
Perhaps this is not related, but I can build (3d printer) Marlin firmware without an issue....

reefwarrior, 2020-09-05:
> #2343 was closed via [e1b643f](https://github.com/xoseperez/espurna/commit/e1b643fd0021692cf1a247a6e527068d08cdaac2)
> have you 'pull'ed from the repo again to sync it?
> 
> > PermissionError: [WinError 5] Access is denied: 'C:\Users\Tinus Blaauw.platformio.cache\tmp\pkg-installing-i2fuchvm.git\objects\pack\pack-1e52effe1f9a5937ec9554213076ce1b1e541f6a.idx':
> > File "C:\users\tinus blaauw.platformio\penv\lib\site-packages\platformio\builder\main.py", line 169:
> 
> does not tell me anything useful :/
> if you happen to run vscode, try again with `pio.exe` directly? (`pio.exe run -e $PIO_ENVIRONMENT_NAME`)

Sorry haven't answered - yes I have pulled the latest.

mcspr, 2020-09-05:
WinError 5 issue is assumed to be fixed in PIO:
https://github.com/platformio/platformio-core/issues/3328

idk about Marlin differences, seems almost the same.
https://github.com/xoseperez/espurna/blob/f6ec2422a49f303658cd2d96fb6c3d14cab861a1/code/platformio.ini#L129
can sometimes become
```
https://github.com/me-no-dev/ESPAsyncTCP/archive/7e9ed22.zip
```

but, if WIFISUPPORT is enabled, I see the same `git` clones (which are supposedly the issue with Windows):
https://github.com/MarlinFirmware/Marlin/blob/9913333c33d234264a9072630e8b5ffbf07bad7e/platformio.ini#L357-L361

reefwarrior, 2020-09-05:
OK, SOMEHOW, and I still don't know how, I got this to build. There was a dependency that it didn't pull (a library item), which I installed manually... SPIFlash or something like that it is called. So I will TRY and see if I can get the new firmware uploaded....

mcspr, 2020-09-09:
SPIFlash was the specific lib that always failed? We don't list it directly, but it comes from rfm69. Does removing rfm69 line from `lib_deps = ...' in the platformio.ini fix the issue?
Libraries are installed per env, have you tried building with another, does it fail?

btw pls also ping this issue https://github.com/platformio/platformio-core/issues/3479 if it breaks reliably

Bmooij, 2020-09-09:
The issue is due to the vscode git watch.
vscode "locks" .git folders

As a workaround you could disable the vscode git watch in settings.json:
```json
{
    "git.enabled": false
}
```

tita2000, 2020-09-10:
Hi, where to find the settings.json file? Thanks

mcspr, 2020-09-10:
@tita2000 
See `code/.vscode/settings.json`
You can use File > Preferences from the VSCode menu

tita2000, 2020-09-11:
Ok, I disabled git but still get the following error

`Processing bw_ss4_2 (platform: espressif8266@2.6.2; board: esp01_1m; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 (2.6.2) > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.0 (2.7.4)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.20800.0 (2.8.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing git+https://github.com/madpilot/mDNSResolver#4cfcda1
git version 2.24.0.windows.2
Cloning into 'C:\Users\ttamm\.platformio\.cache\tmp\pkg-installing-wtm4bv7k'...
HEAD is now at 4cfcda1 Print out correct AP name
PermissionError: [WinError 5] Zugriff verweigert: 'C:\\Users\\tx\\.platformio\\.cache\\tmp\\pkg-installing-wtm4bv7k\\.git\\objects\\pack\\pack-1e52effe1f9a5937ec9554213076ce1b1e541f6a.idx':
  File "C:\users\tx\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 169:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\tx\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\tx\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\users\tx\.platformio\platforms\espressif8266\builder\main.py", line 238:
    target_elf = env.BuildProgram()
  File "C:\Users\tx\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\users\tx\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\tx\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\users\tx\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\tx\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\users\tx\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1062:
    project.install_dependencies()
  File "C:\users\tx\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 898:
    lm.install(spec)
  File "c:\users\tx\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 49:
    spec, silent=silent, skip_dependencies=skip_dependencies, force=force
  File "c:\users\tx\.platformio\penv\lib\site-packages\platformio\package\manager\library.py", line 86:
    force=force,
  File "c:\users\tx\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 97:
    pkg = self.install_from_url(spec.url, spec, silent=silent)
  File "c:\users\tx\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 152:
    return self._install_tmp_pkg(pkg_item)
  File "c:\users\tx\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 241:
    shutil.move(tmp_pkg.path, dst_pkg.path)
  File "C:\Users\tx\.platformio\python3\lib\shutil.py", line 578:
    rmtree(src)
  File "C:\Users\tx\.platformio\python3\lib\shutil.py", line 516:
    return _rmtree_unsafe(path, onerror)
  File "C:\Users\tx\.platformio\python3\lib\shutil.py", line 395:
    _rmtree_unsafe(fullname, onerror)
  File "C:\Users\tx\.platformio\python3\lib\shutil.py", line 395:
    _rmtree_unsafe(fullname, onerror)
  File "C:\Users\tx\.platformio\python3\lib\shutil.py", line 395:
    _rmtree_unsafe(fullname, onerror)
  File "C:\Users\tx\.platformio\python3\lib\shutil.py", line 400:
    onerror(os.unlink, fullname, sys.exc_info())
  File "C:\Users\tx\.platformio\python3\lib\shutil.py", line 398:
    os.unlink(fullname)
=========================================================================================== [FAILED] Took 4.89 seconds ===========================================================================================

Environment    Status    Duration
-------------  --------  ------------
bw_ss4_2       FAILED    00:00:04.894
===================================================================================== 1 failed, 0 succeeded in 00:00:04.894 ===================================================================================== 
The terminal process "C:\Users\tx\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.`

I already tried to reinstall complete ide, but without success.
Thanks for your help. Regards Timo

mcspr, 2020-09-12:
Per https://github.com/platformio/platformio-core/issues/3664#issuecomment-691541518
Things should work after `pio upgrade --dev` / when next version 5.0.2 is released

closed 2020-09-12 by mcspr