xoseperez/espurna #2346: OTA: factory reset after upgrade

mcspr, 2020-09-08:
> This is not totally true, it makes sense when we set a hardcoded wifi and mqtt configuration, although it is true that for most use cases it may not make sense
> 
> https://github.com/xoseperez/espurna/blob/53aa1d109e41cd8ec06291ba2747157be128450b/code/espurna/config/general.h#L293
> 
> https://github.com/xoseperez/espurna/blob/53aa1d109e41cd8ec06291ba2747157be128450b/code/espurna/config/general.h#L297
> 
> https://github.com/xoseperez/espurna/blob/53aa1d109e41cd8ec06291ba2747157be128450b/code/espurna/config/general.h#L569

i agree with your statement. i hard code wifi with other details & distribute using nofuss. need a simple way to "factory reset" the device after the new firmware is updated.  i hope this request will be re-considered. @xoseperez

_Originally posted by @sfilinto in https://github.com/xoseperez/espurna/issues/912#issuecomment-688742491_

*pls don't use 2y old issues :>*

mcspr, 2020-09-08:
There are a lot of moving parts here and I'd agree with the previous issue answer that we don't generally want to do this.

What you propose (if I understand correctly), is to delete all local settings and instead use the ones in the new .bin? Optional, or otherwise.
- we need to ensure we *don't* handle settings flash sector rotation, since we don't want those settings anyway
- there is no communication between bootloader & Updater from Arduino Core to ensure we *actually* upgraded, erasing should probably happen on the very next boot

`resetSettings()` is the function you want to do the actual reset, which can be called after `eepromSetup()` function is done. Basically:
```diff
diff --git a/code/espurna/settings.cpp b/code/espurna/settings.cpp
index bb4cad60..2751afcf 100644
--- a/code/espurna/settings.cpp
+++ b/code/espurna/settings.cpp
@@ -411,6 +411,15 @@ void settingsProcessConfig(const settings_cfg_list_t& config, settings_filter_t

 void settingsSetup() {

+    switch (customResetReason()) {
+    case CUSTOM_RESET_OTA:
+    case CUSTOM_RESET_NOFUSS:
+    case CUSTOM_RESET_UPGRADE:
+        resetSettings();
+        customResetReason(0);
+        break;
+    }
+
     terminalRegisterCommand(F("CONFIG"), [](const terminal::CommandContext& ctx) {
         // TODO: enough of a buffer?
         DynamicJsonBuffer jsonBuffer(1024);
```

Which does the *thing*, but it always triggers on OTA. And it never handles the settings rotation issue mentioned above.

pniaps, 2020-10-05:
What would be necessary is not to factory reset after every OTA, is a way of doing the factory reset remotely by MQTT, the same way we can do a reboot now.

Something similar to 

```
bool rpcHandleAction(const String& action) {
    bool result = false;
    if (action.equals("reboot")) {
        result = true;
        schedule_function([]() {
            deferredReset(100, CUSTOM_RESET_RPC);
        });
    } else if (action.equals("factory_reset")) {
        result = true;
        schedule_function([]() {
            resetSettings();
            deferredReset(100, CUSTOM_RESET_MQTT);
        });
    } else if (action.equals("heartbeat")) {
        result = true;
        schedule_function(heartbeat);
    }
    return result;
}
```

This way, we can manually make a factory reset after upgrade if we know default settings have changed and need to factory reset.

Do you mean this @sfilinto?

BTW, @mcspr how can I enter a diff in the comment, like your previous message.

mcspr, 2020-10-05:
But what would be the approach for nofuss, since the device itself initiates the upgrade?

I'll also answer my own question about the versioning - we can always store some 'settings' keyvalue with the built timestamp and compare it with the one in the .bin. And, another pair which can control the 'policy' for timestamp upgrade, whether to reset settings or do something else (or nothing at all)

@pniaps [if the code block has 'diff' after the top three backticks, it will try to color the contents](https://github.github.com/gfm/#info-string) (i.e. \`\`\`diff ...text... \`\`\`)

Also note of the:
https://github.com/xoseperez/espurna/blob/aebefdf941b30d3db57c1dda4f2a2f7ea8e9d8c4/code/espurna/config/general.h#L183
https://github.com/xoseperez/espurna/blob/aebefdf941b30d3db57c1dda4f2a2f7ea8e9d8c4/code/espurna/config/general.h#L1207

sfilinto, 2020-10-05:
> What would be necessary is not to factory reset after every OTA, is a way of doing the factory reset remotely by MQTT, the same way we can do a reboot now.
> 
> Something similar to
> 
> ```c++
> bool rpcHandleAction(const String& action) {
>     bool result = false;
>     if (action.equals("reboot")) {
>         result = true;
>         schedule_function([]() {
>             deferredReset(100, CUSTOM_RESET_RPC);
>         });
>     } else if (action.equals("factory_reset")) {
>         result = true;
>         schedule_function([]() {
>             resetSettings();
>             deferredReset(100, CUSTOM_RESET_MQTT);
>         });
>     } else if (action.equals("heartbeat")) {
>         result = true;
>         schedule_function(heartbeat);
>     }
>     return result;
> }
> ```
> 
> This way, we can manually make a factory reset after upgrade if we know default settings have changed and need to factory reset.
> 
> Do you mean this @sfilinto?
> 
> BTW, @mcspr how can I enter a diff in the comment, like your previous message.

yes, for the scenario i am referring to, a remotely triggered (via MQTT) factory reset similar to the reboot functionality would be great.