xoseperez/espurna #2354: NeoCoolcam powerplug not getting power after flashing Espurna firmware

erwinjong, 2020-09-16:
firmware 1.14.1 is now running and seems to work fine but the plug is not giving power. first flash was done to firmware 1.113.5 and then upgraded to 1.14.1.

is there configuration needed to make this work or did the firmware brake something ?

erwinjong, 2020-09-16:
Here some more info, did i forgot something regarding flashing the firmwarefile for this neocoolcam wifi plug ?

[644572] [MAIN] ESPURNA 1.14.1 (62ad7da3)
[644573] [MAIN] CPU chip ID: 0x9614DA
[644576] [MAIN] CPU frequency: 80 MHz
[644580] [MAIN] SDK version: 1.5.3(aec24ac9)
[644583] [MAIN] Core version: 2.3.0
[644589] [MAIN] Core revision: 9826c6d
[644590] [MAIN] Build time: 1577762597
[644594] 
[644595] [MAIN] Flash chip ID: 0x146085
[644600] [MAIN] Flash speed: 40000000 Hz
[644601] [MAIN] Flash mode: DOUT
[644605] 
[644606] [MAIN] Flash size (CHIP)   :  1048576 bytes /  256 sectors (   0 to  255)
[644616] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[644620] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[644629] [MAIN] Firmware size       :   484800 bytes /  119 sectors (   1 to  119)
[644634] [MAIN] Max OTA size        :   536576 bytes /  131 sectors ( 120 to  250)
[644641] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[644648] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[644655] 
[644656] [MAIN] EEPROM sectors: 251, 250
[644663] [MAIN] EEPROM current: 250
[644663] 
[644664] [MAIN] EEPROM:  4096 bytes initially |   408 bytes used ( 9%) |  3688 bytes free (90%)
[644674] [MAIN] Heap  : 34688 bytes initially | 14528 bytes used (41%) | 20160 bytes free (58%)
[644681] [MAIN] Stack :  4096 bytes initially |  1320 bytes used (32%) |  2776 bytes free (67%)
[644689] 
[644690] [MAIN] Boot version: 31
[644692] [MAIN] Boot mode: 1
[644699] [MAIN] Last reset reason: External System
[644700] [MAIN] Last reset info: Fatal exception:0 flag:6 (EXT_SYS_RST) epc1:0x00000000 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
[644714] 
[644715] [MAIN] Board: NEO_COOLCAM_NAS_WR01W
[644718] [MAIN] Support: ALEXA API BROKER BUTTON DEBUG_SERIAL DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT LED MDNS_SERVER MQTT NTP SCHEDULER TELNET TERMINAL THINGSPEAK WEB 
[644733] [MAIN] OTA: ARDUINO ASYNCTCP WEB 
[644737] [MAIN] WebUI image: SMALL
[644740] 
[644744] [MAIN] Firmware MD5: 84b583eaf03287bb0b9ad9beaf3b49c9
[644747] [MAIN] Power: 3164 mV
[644750] [MAIN] Power saving delay value: 10 ms
[644753] 

---8<-------

[700249] [MAIN] Heap  : 34688 bytes initially | 13728 bytes used (39%) | 20960 bytes free (60%)
[700249] [MAIN] Power: 3166 mV
[700251] [MAIN] Time: 2020-09-16 17:35:58
[704301] [WEBSOCKET] Requested action: dbgcmd
[704303] Printing all GPIO pins:
[704303] GPIO input pin 0 is HIGH
[704303] GPIO input pin 1 is LOW
[704304] GPIO input pin 2 is HIGH
[704306] GPIO input pin 3 is LOW
[704309] GPIO output pin 4 is HIGH
[704312] GPIO input pin 5 is LOW
[704315] GPIO output pin 12 is HIGH
[704321] GPIO input pin 13 is HIGH
[704321] GPIO input pin 14 is LOW
[704324] GPIO input pin 15 is LOW
[704327] +OK

switch:
  - platform: mqtt
    name: NeoCoolcam
    state_topic: 
    command_topic: 
    payload_on: 1
    payload_off: 0
    availability_topic: 
    payload_available: 
    payload_not_available:

mcspr, 2020-09-16:
You can't control the device in the WebUI or via the Homeassistant config you provided below?
If the 2nd - enable MQTT and set a correct server. There's a small 'bug' with the display here - topics are never shown unless MQTT integration is enabled

erwinjong, 2020-09-16:
perhaps anyone knows what the issue is ? maybe it has to with the plug i use. I have the NAS-WR01ZE 16A plug which perhaps uses other GPIO's so perhaps i need another firmware then. Espurna shows: Switch #0 ( GPIO12)

I do like Espurna since the firmware works fine besides this issue.

NEO device is connected as (NAS-OGCZ-WIFI-T 20180125 v3)
LED = GPIO4 (D2 NodeMCU)
RELAIS = GPIO12 (D6 ModeMCU)
BUTTON = GPIO13 (D7 NodeMCU)
NEO device is connected as (NAS-OGCZ-16A-T 20190428 v5)
LED = GPIO13
RELAIS = GPIO14
BUTTON = GPIO0

[image]

erwinjong, 2020-09-16:
Hello Max,

Yes indeed, switch relay is not doing anything. But i think i see the issue
perhaps. Espurna shows  Switch #0 (GPIO12)  as switch so i think i have
firmware for another Neo Coolcam plug. Mine is for 2300V/16A so i think
that is the issue.

Only question is, if there is another flash file or how i can get a
modified one for the     NAS-OGCZ-16A-T 20190428 v5 ?

Thanks for helping by the way ! Much appreciated. I like the firmware and
because of the more advantages i wanted to use Espurna instead of the other
ones.

Regards, Erwin de Jong

NEO device is connected as (NAS-OGCZ-WIFI-T 20180125 v3)

   - LED = GPIO4 (D2 NodeMCU)
   - RELAIS = GPIO12 (D6 ModeMCU)
   - BUTTON = GPIO13 (D7 NodeMCU)

<https://github.com/xoseperez/espurna/wiki/Hardware-NEO-COOLCAM-NAS-WR01W#neo-device-is-connected-as-nas-ogcz-16a-t-20190428-v5>NEO
device is connected as (NAS-OGCZ-16A-T 20190428 v5)

   - LED = GPIO13
   - RELAIS = GPIO14
   - BUTTON = GPIO0

erwinjong, 2020-09-16:
Ok, i know for sure now, i have a different plug from the same brand.

Can i still use Espurna some how ?

Regards, Erwin de Jong

https://templates.blakadder.com/neo_coolcam_NAS-WR01W-16A.html

GPIO #Component
GPIO00 Button1
GPIO01 None
GPIO02 None
GPIO03 None
GPIO04 BL0937 CF
GPIO05 HLWBL CF1
GPIO09 None
GPIO10 None
GPIO12 HLWBL SELi
GPIO13 Led1i
GPIO14 Relay1
GPIO15 None
GPIO16 None
FLAG None

erwinjong, 2020-09-16:
Ok the issue is clear now, I have the NEO Coolcam NAS-WR01W 16A Power plug instead of the NEO Coolcam NAS-WR01W Plug

The 16A has these GPIO like this ? Is it possible to modify/compile the firmware bin-file to these GPIO settings perhaps ?

GPIO # | Component
-- | --
GPIO00 | Button1
GPIO01 | None
GPIO02 | None
GPIO03 | None
GPIO04 | BL0937 CF
GPIO05 | HLWBL CF1
GPIO09 | None
GPIO10 | None
GPIO12 | HLWBL SELi
GPIO13 | Led1i
GPIO14 | Relay1
GPIO15 | None
GPIO16 | None
FLAG | None

mcspr, 2020-09-16:
There are at least 3 similar configs. For example:
https://github.com/xoseperez/espurna/blob/1d895d98b801baf5cba6b2c876589ad63e177a50/code/espurna/config/hardware.h#L4627-L4665
Might need power meter calibration though, idk about the ratio values there.

Check out the kogan .bin

erwinjong, 2020-09-17:
Ok, i can see that config has the right GPIO settings, i only cannot find a bin file that looks like kogan.bin on https://github.com/xoseperez/espurna/releases.

Is there somewhere a kogan.bin i can use  to flash ? or another way i can change the config to the proper
Also one other question if i can flash any other firmware then espurna if i cannot find a fix for this plug ?

thanks in advance

mcspr, 2020-09-17:
Hm. There is also kmc-70011.bin
https://github.com/xoseperez/espurna/blob/1d895d98b801baf5cba6b2c876589ad63e177a50/code/espurna/config/hardware.h#L2049-L2077

What I was referring earlier is only available in the development version at
https://github.com/mcspr/espurna-nightly-builder/releases

erwinjong, 2020-09-17:
ah ok, ik found them.
so either espurna-1.15.0-dev.nightly20200916+git1d895d98-kogan-smarter-home-plug-w-pow.bin or espurna-1.15.0-dev.nightly20200916+git1d895d98-kmc-70011.bin should maybe work for this plug then ?

i will flash kogan first later and report back how it works :)

erwinjong, 2020-09-17:
ok, the kogan seems to work fine. Did use espurna-1.14.2-dev.nightly20200121+git495fa18a-kogan-smarter-home-plug-w-pow.bin since that one is 495Kb and fits through the webinterface OTA.

Looking at the hardware.h content the kogan version seems to be the better one so i will stick to this one.

Thanks for helping out !
maybe for the future these GPIO settings could be included in the firmware for NeoCoolcam for rookies like me.

erwinjong, 2020-10-18:
hello, i have two questions, one is about the kogan firmware, it works great (switch and sensor) for some weeks and then gets into issues and looses its configuration.

The other one is i have the normal (10A) NAS-WR01W NeoCoolcam wifi plug which work fine regarding the switch but not seeing sensors and thus no power information.

Tried the 1.14.1 firmware but also some newer firmwares (now runs 1.14.2-dev) so i wonder if the two plugs i have running with Espurna have a different GPIO for that sensor or that both have a failing sensor for power management.

[561302] GPIO input pin 0 is HIGH
[561302] GPIO input pin 1 is LOW
[561302] GPIO input pin 2 is HIGH
[561305] GPIO input pin 3 is LOW
[561308] GPIO output pin 4 is LOW
[561311] GPIO input pin 5 is HIGH
[561313] GPIO output pin 12 is HIGH
[561319] GPIO input pin 13 is HIGH
[561320] GPIO input pin 14 is HIGH
[561323] GPIO input pin 15 is LOW

hapklaar, 2022-10-10:
Is this firmware supported on the NAS-WR01ZE which is a Z-Wave Plus device afaik.