xoseperez/espurna #2356: RF Bridge Direct Mode Hack - Continous sending signal

alceasan, 2020-09-16:
Related with issue #2252 , already closed but not solved.

After flashing espurna on Sonoff Bridge RF using direct mode hack, whenever I send a signal with it, it keeps sending signals and blocking any other 433Mhz transmissions. It's a real problem because you can't use any other 433Mhz devices, my garage door RF opened doesn't work (nor the nearest neighbors).

The only solution I have found so far is shutting down a few seconds and up again every time I send a signal with it. I first tried to reboot it, but it only stops when cutting power completely.

mcspr, 2020-09-16:
Software does not hold the pin, I was just checking RCSwitch integration and that the RF transmitter works properly because of https://gitter.im/tinkerman-cat/espurna?at=5f5e7309ea62d70e650d36c4
But, that is just with a dev board, not the rfbridge. And in the end it was invalid wiring that caused efm8b11 to try to read terminal output.

Something with the HW mod side, perhaps some additional damage to the traces?
@wildwiz

wildwiz, 2020-09-18:
here are the possible cases I see:

**hardware: standard firmware: modified**
the transmitter is still controlled by the efm8b11 which the mod firmware does not drive.
so, nothing is transmitted and nothing is received

**hardware: modified firmware: standard**
the transmitter is driven by esp8266 #define RFB_TX_PIN 5 but there is nobody driving that.
I don't know what is the behavior in this case, it depends on how the pin is configured by default.
If, as I think, it is configured as input (sensible safe default), again, nothing can be transmitted nor received.
@mcspr can we assume hat the unused pins are Hi-Z ? otherwise it might s reimply be the wrong firmware

**hardware modified improperly**
an unclean cut may leave any of the lines either shorted to adjacent copper or remain connected where it should not
be sure that the cut is effective and no metal remains bridging the two sides of the cut or the cut conductor with an adjacent ground/power/line

In any case, if you want to further troubleshoot without actually transmitting (which is a responsible thing to do),
just remove the resistor/jumper attached to R13 and ensure the cut nearby is clean.
Under these circumstances, no RF can be transmitted, and you can measure the voltage on the wire you raised from R13 to see if it 'wants to transmit' or not. You might even add a led (with a resistor!) to see it visually
(an oscilloscope is better if you have one of course)

mcspr, 2020-09-19:
> can we assume hat the unused pins are Hi-Z ? otherwise it might simply be the wrong firmware

Arduino Core init sequence includes `pinMode(pin, INPUT)` on all available pins, so yes
https://github.com/esp8266/Arduino/blob/2.7.4/cores/esp8266/core_esp8266_wiring_digital.cpp#L240-L254
(note that we can override this behaviour by adding our own `void __resetPins()` somewhere in the .cpp)

closed 2020-12-19 by stale[bot]