xoseperez/espurna #2359: How to communicate with other board using this framework via UART?

vdj69, 2020-09-21:
Hi there!

Special thanks to all the contributors for this awesome work. I want to use nodemcu to communicate to another controller via UART which doesn't have wifi. I basically want to control the board via sending the MQTT command and receive the response via MQTT.

mcspr, 2020-09-23:
Take a look at any other modules using `mqttSubscribe` / `mqttSend`?

It seems too broad of a question, I'd suggest to try to look up any Arduino examples and put the code in the loop function, see https://github.com/xoseperez/espurna/wiki/3rd-Party-Plugins.
Important to note that `MQTT` incoming messages callback executes in a different 'thread' from the main `loop()`. Meaning, when you `yield()` in the loop code (`DEBUG_MSG...`, `delay(ms);`, etc.), you might receive MQTT message and you should not change things that loop might depend on / want in a certain state.

tsymbaliuk, 2020-10-13:
@vdj69 , I'm doing exactly what you want to do. What you need is to build software with UART_TO_MQTT support (there are no releases with such functionality yet). After that, you'll be able to receive and send messages from specific MQTT topics through serial.

closed 2022-04-17 by stale[bot]