xoseperez/espurna #2365: Is there need of a timeout for MQTT connecting status?

flucasro, 2020-09-24:
In 1.14.1  I was experiencing recurrent blocking issues when trying to connect to MQTT server

Entering in "connecting" status, and never progressing (Disconnection or full connection)

I implemented some workaround (a timeout, for entering into disconnection). I have ported it to dev branch, 
Although I have not done extensive testing, if it is still needed in the dev branch, I do not see any major change to prevent the case of an endless connecting status.

Obviously there must be some issue in the MQTT library and its event handling.... but I do not know if it is even possible to ensure that this library will provide 100% of disconnection callbacks... Therefore this is why I have on my devices the hardcoded timeout

Proposed change:

 code/espurna/mqtt.cpp | 4 ++++
 1 file changed, 4 insertions(+)

diff --git a/code/espurna/mqtt.cpp b/code/espurna/mqtt.cpp
index 62e0fdb6..584b5096 100644
--- a/code/espurna/mqtt.cpp
+++ b/code/espurna/mqtt.cpp
@@ -70,6 +70,7 @@ Updated secure client support by Niek van der Maas < mail at niekvandermaas dot
 #endif // MQTT_LIBRARY == MQTT_ASYNCMQTTCLIENT
 
 

> +unsigned long _mqtt_last_connecting = 0;
>  unsigned long _mqtt_last_connection = 0;
>  AsyncClientState _mqtt_state = AsyncClientState::Disconnected;
>  bool _mqtt_skip_messages = false;
> @@ -981,6 +982,8 @@ void _mqttConnect() {
>      // Do not connect if disabled
>      if (!_mqtt_enabled) return;
>  
> +    if ((_mqtt_state == AsyncClientState::Connecting) && (millis() - _mqtt_last_connecting > _mqtt_reconnect_delay)) {_mqtt_state = AsyncClientState::Disconnected;};
> +
>      // Do not connect if already connected or still trying to connect
>      if (_mqtt.connected() || (_mqtt_state != AsyncClientState::Disconnected)) return;
>  
> @@ -1006,6 +1009,7 @@ void _mqttConnect() {
>      DEBUG_MSG_P(PSTR("[MQTT] Will topic: %s\n"), _mqtt_will.c_str());
>  
>      _mqtt_state = AsyncClientState::Connecting;
> +    _mqtt_last_connecting = millis();
>  
>      _mqtt_skip_messages = (_mqtt_skip_time > 0);
>  
>

mcspr, 2020-09-24:
Would you mind also checking what `netstat` shows related to the target server when this happens?

What value is `mqttKeep` / `MQTT_KEEPALIVE` for the device?
As far as I remember, async-mqtt-client may not reliably check for the keepalive, in case you have it set as ~10min like the default in the 1.14.1. Meaning, if network dies at exactly the time it tries to connect, you'd get a hanging mqtt client.
But, you will eventually get disconnected in ~10-15 minutes :) Not that it helps though.
*(and there is also an issue while still connected, low heartbeat & high ping might also fill the internal buffer, so the ping won't even be sent)*

mcspr, 2020-09-25:
Also, see the previous issue https://github.com/xoseperez/espurna/issues/2112#issuecomment-575878757 in the issue gone stale b/c of no comments after mine. ESPAsyncTCP library has mechanism similar to the MQTT keepalive, counting response acks and time it takes to get them (which we never receive in this case).

So we need to adjust both the low-level lib and mqtt approach of talking to it. It feels like the timer stuff should happen in the library, not in ours code

flucasro, 2020-09-25:
Seems that we have irreconcilable approaches about how to handle things...

I fully disagree with this statement "So we need to adjust both the low-level lib and mqtt approach of talking to it. It feels like the timer stuff should happen in the library, not in ours code" conceptually and as a way to handle the problems.

I will keep my own fork.

mcspr, 2020-09-25:
Have you tried any of the patches for the libraries?

mcspr, 2020-09-25:
ol, ok, nvm :) deleting the account is also an option

Let me rephrase - it is a valid patch, but it *does not solve the underlying issue*. We simply add another crutch. Which seems simple, at a glance, but we add another level of complexity we don't actually want