xoseperez/espurna #2368: Cannot compile with counter sensor option enabled

Dilepa, 2020-09-25:
Hi,
Trying to compile firmware for d1 mini with counter sensor option in custom.h file.

I get these error:
c:/users/dileepar/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\my-d1mini_test_sensors\src\sensor.cpp.o:(.iram.text.sensors/EventSensor.h.188.2+0x4): undefined reference to `__atomic_fetch_add_4'
c:/users/dileepar/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\my-d1mini_test_sensors\src\sensor.cpp.o: in function `_event_sensor_isr(unsigned char)':
c:\users\dileepar\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits/atomic_base.h:474: undefined reference to `__atomic_fetch_add_4'

Any idea why am getting the errors? I am on dev branch and cusom.h file:

    //Enable Events
    #define EVENTS_SUPPORT              1       // Do not build with counter support by default

    //Event number 1
    #define EVENTS1_PIN                 13

closed 2020-09-26 by mcspr

mcspr, 2020-09-27:
Should be fixed. I'm able to read dummy pwm, but I have not checked if it has any issues with higher frequencies
(which would be kind of expected, since we use CPU to do this stuff)

Can check out the changes if there is any difference between atomic / non-atomic commits?
- 8252f2ae1899f89c974b787fdeb074ab96858116 what I originally had done to fix this
- e84fa98941cbbab4f377c8d43e9bc9cdd54f35a2 slightly different reading technique
- 723636ae21a7186d247d7a55097cb20c6e9d6442 reverted to be closer to the 1.14.1 to with less code, but now this stops other GPIO interrupts for firing while 'reading' the value

Dilepa, 2020-09-27:
Hi,
Thanks, This 8252f2a works and i can compile without issues, am a total newbie not sure about atomic/non-atomic commits!
Sorry.

mcspr, 2020-09-27:
I mean, could you check out the 2 other commits?

atomic / non atomic refers to the internal stuff, sorry for confusion! I meant to point those are slightly different implementations which might work differently

Dilepa, 2020-09-28:
@mcspr ,
Am really sorry, am not really familiar with the workings of github. How can you check different commits? I really think am not the right person to continue this.

mcspr, 2020-09-28:
Three commits in question are part of the dev branch and you should be able to use them after pulling changes and running `git checkout commit-hash -- EventSensor.h` in the `code/espurna/sensors` directory
(which will replace the existing file you have - https://www.git-tower.com/learn/git/faq/detached-head-when-checkout-commit/#understanding-how-checkout-works - just use `git checkout EventSensor.h` afterwards to restore it to the latest version)

Dilepa, 2020-09-29:
@mcspr ,

Checked all 3 commits, Can compile and it works.

The problem is the events are not sent straightaway when the GPIO is triggered. I get the total count but not the event.

00037] [MQTT] Sending ESPURNA-78D9CD/rssi => -39 (PID 1)
[300043] [MQTT] Sending ESPURNA-78D9CD/uptime => 300 (PID 1)
[300054] [MQTT] Sending ESPURNA-78D9CD/freeheap => 21576 (PID 1)
[300057] [MQTT] Sending ESPURNA-78D9CD/vcc => 2984 (PID 1)
[300058] [MQTT] Sending ESPURNA-78D9CD/status => 1 (PID 1)
[300064] [MQTT] Sending ESPURNA-78D9CD/loadavg => 1 (PID 1)
[302788] [MQTT] Sending ESPURNA-78D9CD/digital => 1 (PID 1)
[302790] [MQTT] Sending ESPURNA-78D9CD/count => 7 (PID 1)
[302791] [MQTT] Sending ESPURNA-78D9CD/event => 0 (PID 1)

mcspr, 2020-09-29:
Event Sensor triggers were part of the previous release, but right now there is a slightly different mechanism to make this happen (which I believe this is not properly mentioned in the docs... besides, EventSensor should be probably named CounterSensor). Also note that 'event' does not correspond to the digital state of the pin, but a 'count > 0' translated into either 1 or 0

By default, sensor MQTT report happens after a certain number of readings, but this can be changed for any 'magnitude' individually. For example, by using `set digitalMaxDelta0 1` and `reload` you will force the ESPURNA-78D9CD/digital message each time the value changes from 0 to 1 and back, after each reading.

commit 8252f2ae (2020-09-26): sns: explicitly load before storing new counter value

resolve #2368

- Fixes "undefined reference to `__atomic_fetch_add_4'"
  Notably, std::atomic_fetch_add(&counter, 1) would also trigger this.
  std::atomic<T>::operator+= does not allow in-place change, since it
  returns T instead of T&
- Another work reduction in the isr, treat 'value' as `counter > 0`
  instead of relying on the notion that we can sometimes read the value
  with very low counter frequency
- Fix duplicated initializations