xoseperez/espurna #2372: sonoff 4ch pro R3 version

thereal65man, 2020-09-28:
*Before creating a new issue please check that you have:*

* *searched the existing [issues](https://github.com/xoseperez/espurna/issues) (both open and closed)*
* *searched the [wiki](https://github.com/xoseperez/espurna/wiki)*
* *asked for help in the [chat](https://gitter.im/tinkerman-cat/espurna)*
* *done the previous things again :)*

*Fulfilling this template will help developers and contributors help you. Try to be as specific and extensive as possible. If the information provided is not enough the issue will likely be closed.*

*You can now remove this line and the above ones. Text in italic is meant to be replaced by your own words. If any of the sections below are not relevant to the issue (for instance, the screenshots) then you can delete them.*

**Question**
*A clear and concise description of what the problem/doubt is.*

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by typing `info` via serial, terminal or in the debug tab in the web UI. The relevant information is that surrounded by the scissors-cut lines (`---8<-------`).*
*If you cannot get this info from the device, please answer this questions:*
* *Arduino Core version*
* *ESPurna version*
* *Flash mode*
* *Device brand, model and version*

**Tools used**
* *Desktop operating system*
* *Browser & version*
* *IDE & version*
* *Compiler & version (if not embedded in IDE)*

**Additional context**
*Add any other context about the problem here.*

thereal65man, 2020-09-28:
I would ask if the espurna firmware version for the new sonoff 4ch pro R3 is the same that R2. Thank you for the answer.

ColinShorts, 2020-09-28:
It would appear to work the same as the original 4ch, have a look at [this link](https://notenoughtech.com/sonoff/flashing-tasmota-on-4ch-r3-4chpro-r3/) explaining how to setup tasmota. It mentioned that the config is the same.

thereal65man, 2020-09-28:
thank you Colin! I saw the link, i hope espurna for R2 work in R3...

ColinShorts, 2020-09-28:
It should hopefully be exactly the same for the original version, R2, and RE. The pro is different from the regular version, but there is no distinction between Original and R2 - see hardware.h / https://github.com/xoseperez/espurna/wiki/Hardware-Itead-Sonoff-4CH-Pro---R2

There may be physical differences on the board - if there are it would be worth taking pictures and having the wiki updated.

thereal65man, 2020-09-28:
... no anyone buy the new R3 ch4 PRO and try to flash it with espurna?...

thereal65man, 2020-10-06:
Ok! I just do it! I flashed espurna on sonoff 4ch pro R3! no problems friends, all ok! thank you for the support! ciao!

closed 2020-10-06 by mcspr