xoseperez/espurna #2373: Sonoff T3 reboot avery few minutes

pcornill, 2020-09-28:
Hi,
I just flashed a Sonoff T3 2CH with Espurna 1.14.2-dev and it constantly reboots after few minutes of running.

It seems the reason is: Hardware button but I do not do anything!

LOGs:
 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1392, room 16 
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld
[000272] 

---8<-------

[000272] [MAIN] ESPURNA 1.14.2-dev
[000272] [MAIN] xose.perez@gmail.com
[000273] [MAIN] http://tinkerman.cat

[000273] [MAIN] CPU chip ID: 0xAA04FE
[000276] [MAIN] CPU frequency: 80 MHz
[000279] [MAIN] SDK version: 2.2.2-dev(38a443e)
[000284] [MAIN] Core version: 2.6.3
[000287] [MAIN] Core revision: 3d128e5c
[000290] [MAIN] Build time: 1601289030
[000293] 
[000295] [MAIN] Flash chip ID: 0x144051
[000298] [MAIN] Flash speed: 40000000 Hz
[000301] [MAIN] Flash mode: DOUT
[000304] 
[000305] [MAIN] Flash size (CHIP)   :  1048576 bytes /  256 sectors (   0 to  255)
[000313] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[000320] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[000327] [MAIN] Firmware size       :   568928 bytes /  139 sectors (   1 to  139)
[000334] [MAIN] Max OTA size        :   454656 bytes /  111 sectors ( 140 to  250)
[000341] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[000348] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[000356] 
[000357] [MAIN] EEPROM sectors: 251, 250
[000360] [MAIN] EEPROM current: 250
[000363] 
[000364] [MAIN] EEPROM:  4096 bytes initially |   263 bytes used ( 6%) |  3833 bytes free (93%)
[000373] [MAIN] Heap  : 39464 bytes initially |  5280 bytes used (13%) | 34184 bytes free (86%)
[000381] [MAIN] Stack :  4096 bytes initially |  1232 bytes used (30%) |  2864 bytes free (69%)
[000389] 
[000390] [MAIN] Boot version: 31
[000393] [MAIN] Boot mode: 1
[000395] [MAIN] Last reset reason: Hardware button
[000400] 
[000401] [MAIN] Board: ITEAD_SONOFF_T1_2CH
[000404] [MAIN] Support: ALEXA API BROKER BUTTON DEBUG_SERIAL DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT LED MDNS_SERVER MQTT NTP SCHEDULER TELNET TERMINAL THINGSPEAK WEB 
[000419] [MAIN] OTA: ARDUINO ASYNCTCP WEB 
[000423] [MAIN] WebUI image: SMALL
[000426] 
[000428] [MAIN] Power: 4324 mV
[000430] [MAIN] WiFi Sleep Mode: MODEM
[000433] 

---8<-------

[000445] [TELNET] Listening on port 23
[000447] [WEBSERVER] Webserver running on port 80
[000451] [RELAY] Retrieving mask: 0b10
[000451] [RELAY] Relay #0 boot mode 0
[000452] [RELAY] Relay #1 boot mode 0
[000453] [RELAY] #0 set to OFF
[000456] [RELAY] #1 set to OFF
[000460] [RELAY] Number of relays: 2
[000468] [BUTTON] Number of buttons: 2
[000469] [LED] Number of leds: 1
[000469] [MQTT] AsyncMqttClient, SSL DISABLED, Autoconnect ENABLED, Buffer size 1024 bytes
[000476] [MQTT] Client DISABLED, DISCONNECTED
[000488] [NTP] Startup delay: 5s, Update delay: 3231s
[000490] [NTP] Server: pool.ntp.org, TZ: UTC0
[000493] [THINGSPEAK] Async ENABLED, SSL DISABLED
[000708] [WIFI] Scanning
[001459] [RELAY] Setting relay mask: 0b0
[002901] [WIFI]     BSSID: 8C:F8:13:4E:1E:AC CH:  1 RSSI: -92 SEC: WPA2 SSID: Livebox-1EAC
[002902] [WIFI]     BSSID: 8C:F8:13:4E:1E:AD CH:  1 RSSI: -92 SEC: OPEN SSID: orange
[002906] [WIFI] --> BSSID: D8:9E:61:DA:6E:CD CH:  7 RSSI: -59 SEC: WPA2 SSID: Pantheres
[002914] [WIFI]     BSSID: AC:3A:7A:0D:18:2F CH:  7 RSSI: -62 SEC: WPA2 SSID: 
[002931] [WIFI] Connecting to BSSID: D8:9E:61:DA:6E:CD CH: 07, RSSI: -59, SEC: WPA2, SSID: Pantheres
[005949] [WIFI] Captive portal disabled
[005950] [WIFI] ------------------------------------- MODE STA
[005950] [WIFI] SSID  Pantheres
[005951] [WIFI] IP    192.168.1.170
[005954] [WIFI] MAC   2C:F4:32:AA:04:FE
[005957] [WIFI] GW    192.168.1.1
[005960] [WIFI] DNS   192.168.1.1
[005963] [WIFI] MASK  255.255.255.0
[005966] [WIFI] HOST  http://SonoffTX170.local
[005970] [WIFI] BSSID D8:9E:61:DA:6E:CD
[005974] [WIFI] CH    7
[005976] [WIFI] RSSI  -60
[005978] [WIFI] ----------------------------------------------
[005985] [MDNS] OK
[006987] [MQTT] MQTT brokers found: 0
[006989] [MQTT] Connecting to broker at iot.westeurope.cloudapp.azure.com:1883
[006990] [MQTT] Client ID: ESPURNA-AA04FE
[006991] [MQTT] QoS: 1
[006993] [MQTT] Retain flag: 1
[006995] [MQTT] Keepalive time: 120s
[006999] [MQTT] Will topic: SonoffTX170/status
[007163] [MQTT] Connected!
[007164] [MQTT] Unsubscribing to # (PID 1)
[007165] [MQTT] Subscribing to SonoffTX170/relay/+/set (PID 2)
[007165] [MQTT] Subscribing to SonoffTX170/pulse/+/set (PID 3)
[007170] [MQTT] Subscribing to SonoffTX170/led/+/set (PID 4)
[007175] [MQTT] Subscribing to SonoffTX170/action/set (PID 5)
[007381] [MQTT] Subscribe ACK for PID 2
[007382] [MQTT] Subscribe ACK for PID 3
[007382] [MQTT] Subscribe ACK for PID 4
[007382] [MQTT] Subscribe ACK for PID 5
[011005] [NTP] Server     : pool.ntp.org
[011006] [NTP] Sync Time  : 2020-09-28 09:25:48 (UTC)
[011006] [NTP] UTC Time   : 2020-09-28 09:25:48
[011009] [MQTT] Sending SonoffTX170/app => ESPURNA (PID 6)
[011012] [MQTT] Sending SonoffTX170/version => 1.14.2-dev (PID 7)
[011018] [MQTT] Sending SonoffTX170/board => ITEAD_SONOFF_T1_2CH (PID 8)
[011024] [MQTT] Sending SonoffTX170/host => SonoffTX170 (PID 9)
[011030] [MQTT] Sending SonoffTX170/ssid => Pantheres (PID 10)
[011035] [MQTT] Sending SonoffTX170/ip => 192.168.1.170 (PID 11)
[011041] [MQTT] Sending SonoffTX170/mac => 2C:F4:32:AA:04:FE (PID 12)
[011047] [MQTT] Sending SonoffTX170/rssi => -61 (PID 13)
[011052] [MQTT] Sending SonoffTX170/uptime => 11 (PID 14)
[011057] [MQTT] Sending SonoffTX170/datetime => 2020-09-28 09:25:48 (PID 15)
[011063] [MQTT] Sending SonoffTX170/freeheap => 25672 (PID 16)
[011069] [MQTT] Sending SonoffTX170/relay/0 => 0 (PID 17)
[011074] [MQTT] Sending SonoffTX170/relay/1 => 0 (PID 18)
[011080] [MQTT] Sending SonoffTX170/vcc => 4324 (PID 19)
[011084] [MQTT] Sending SonoffTX170/status => 1 (PID 20)
[011089] [MQTT] Sending SonoffTX170/loadavg => 100 (PID 21)
[011142] [MQTT] Publish ACK for PID 6
[011211] [MQTT] Publish ACK for PID 7
[011441] [MQTT] Publish ACK for PID 8
[011442] [MQTT] Publish ACK for PID 9
[011442] [MQTT] Publish ACK for PID 10
[011442] [MQTT] Publish ACK for PID 11
[011444] [MQTT] Publish ACK for PID 12
[011447] [MQTT] Publish ACK for PID 13
[011451] [MQTT] Publish ACK for PID 14
[011454] [MQTT] Publish ACK for PID 15
[011458] [MQTT] Publish ACK for PID 16
[011461] [MQTT] Publish ACK for PID 17
[011464] [MQTT] Publish ACK for PID 18
[011468] [MQTT] Publish ACK for PID 19
[011471] [MQTT] Publish ACK for PID 20
[011475] [MQTT] Publish ACK for PID 21
[060003] [MAIN] System OK

mcspr, 2020-09-29:
I believe this is a duplicate (?) of #2331? Reset reason refers to the button press

Two things to note though:
- the device might be as stable with 5v or 3v3 as with the AC power, especially with the WiFi ON
- you are using the version the current development version, 1.14.2-dev is quite old at this point. please update!

closed 2020-09-29 by mcspr

pcornill, 2020-09-29:
Thanks for your feedback. I do not find the latest version. Is it 1.14.1?

mcspr, 2020-09-29:
>  I do not find the latest version. Is it 1.14.1?

I meant this ->
```
[000272] [MAIN] ESPURNA 1.14.2-dev
...
[000284] [MAIN] Core version: 2.6.3
...
[000290] [MAIN] Build time: 1601289030
```

Just update to the latest `dev` branch commit via `git pull`, assuming you used the the `git clone ...` originally.

pcornill, 2020-09-29:
Thanks Max, that I understood. My question is generic:
From GIT, I see the latest version (1.14.1) and I use an earlier version 1.14.2-dev right?
Is the 1.15.0-dev version stable enough?

mcspr, 2020-09-29:
Yes, but it is a trick question :) It is very likely to hit some bug with 1.14.1 that is already fixed in the current version, both internal or related to some weird API interactions.

Also note of the esp8266 Core version update in case you are using PlatformIO, you should be able to do OTA in one step by using gzip'ed binaries - https://arduino-esp8266.readthedocs.io/en/latest/ota_updates/readme.html#compression