xoseperez/espurna #2383: NodeMCU with latest Espurna already flashed, how to/where add new devices?

romrp, 2020-10-19:
Hello, sorry for the somewhat simple question, but is there a way to hook more gpio ports to the Espurna on a NodeMCU? I flashed with the "latest" for this device. Please point me to the right direction, since I found none. Thank you, Rom

romrp, 2020-10-19:
> Hello, sorry for the somewhat simple question, but is there a way to hook more gpio ports to the Espurna on a NodeMCU? I flashed with the "latest" for this device. Please point me to the right direction, since I found none. Thank you, Rom

Answering my own question, to add new switches, buttons etc, or you recompile the whole thing with PlatformIO, that took me several hours of configuration, etc, or you edit the JSON commands. To do that you can use the web interface while the nodemcu is on. Go to the Admin area, click on backup. It will download a JSON file. Edit this in any text editor and save. Go back to the web interface Admin area, but this time RESTORE and look for the file you just edit. REBOOT. The syntax is not very clear, mostly you read the hardware.h file.

mcspr, 2020-10-21:
Do you have any additions to the https://github.com/xoseperez/espurna/wiki/PlatformIO process?

JSON file is not the only way to update settings, see https://github.com/xoseperez/espurna/wiki/Terminal (and relevant things in the sidebar). key:values you've seen in the JSON can be retrieved with `get`, changed with `set` or deleted with `del`
Relays cannot be updated through runtime, yet. Buttons can, using `btnGPIO#` syntax. And both are not yet exposed in WebUI besides manual settings interaction.

closed 2020-10-21 by mcspr