xoseperez/espurna #2389: Use DHCP for DNS and/or Change DNS via GUI

techfixpros, 2020-10-27:
**Is your feature request related to a problem? Please describe.**
*DNS is set to 8.8.8.8 by default, with no option to change it via GUI. DNS also does not follow DHCP-supplied value.*

**Describe the solution you'd like**
*Provide a GUI interface to change DNS, and/or use DHCP-supplied value instead of 8.8.8.8.*

**Describe alternatives you've considered**
*Taking a backup, modifying "dns0": to desired value, and restoring from modified backup allows changing value.*

mcspr, 2020-10-27:
What version are you using?
https://github.com/xoseperez/espurna/search?q=8.8.8.8 yields no results, I believe I removed them from both default settings and WebUI some time ago and even 1.14.1 includes this change:
8e7854b04e3d6dfef22b29444218b08dc7425b92
However, you may have gotten it into settings from previous releases, since old WebUI included text-input default value and may have saved it 

Note that unless you are using static IP setting, gateway and dns settings are ignored in favour of dhcp client configuration

techfixpros, 2020-10-27:
I am on 1.14.1 on all modules, but they were initially running a previous version and were upgraded and restored with backups from previous versions. This can be closed as they retained that value from previous builds. Thank you.

closed 2020-10-27 by techfixpros