xoseperez/espurna #2390: Non-intrusive Wattmeter with Espurna

pcornill, 2020-11-01:
Hello,
I am wondering if there is a wattmeter like the Sonoff POW but with a non-intrusive clamp or ring?
Ideally a device that can be inserted with DIN Rail and > 32 Amps
Something [like that](https://www.amazon.com/Single-phase-Electricity-Watt-hour-Household-Consumption/dp/B08397K9CS/ref=sr_1_31?dchild=1&keywords=watt+meter+wifi&qid=1604250632&sr=8-31) but with Espurna:
Thanks to all in advance

krikk, 2020-12-01:
your linked device can possibly be used with espuna, i have a similar device in use with tasmota, so erspurna should be possible also... see [here ](https://github.com/krikk/Hiking-DDS238-2-WIFI-Din-Rail-Energy-Meter-flashing-Tasmota)for details...

davebuk, 2020-12-12:
See my post #2277 and others using the PZEM-004T. https://www.amazon.co.uk/peng-PZEM-004T-Multimeter-Split-Core-Transformer/dp/B0842NWC1Q/

I used a Wemos D1 mini and 240V to 3V converter inside a double pattress box and fitted the current transformer around the consumer unit neutral. You obviously need to know what you are doing with mains electricity and the dangers involved.