xoseperez/espurna #2392: Support for 80 amps smart switch

goRaspy, 2020-11-01:
I am looking for a compatible 80 amp smart switch.

There is several products working with tuya on AliExpress. But I can't find one confirmed to include an esp8266.

Is someone is able to flash this kind of device ? 

Many thanks

krikk, 2020-12-01:
not 80A but only 65A, but perhaps it helps, and it has an esp8266 because i have it working with tasmota: https://github.com/krikk/Hiking-DDS238-2-WIFI-Din-Rail-Energy-Meter-flashing-Tasmota

timmeh87, 2020-12-24:
80 amps equates to quite a large relay, that is an entire small house worth of 220V AC current, unless its a car, which is a common use for 80amp, 12v relays. (These cannot be used for AC though). Around this size things start to be called "contactors" instead of relays. The coil of such a relay alone might require 5 or 10 watts. You are probably going to have to find an 'industrial' device to get that kind of rating. The parts I can bring up in a quick search cost $30 for just the naked relay.  It might be possible for you to get "small" smart relay and use that small relay to trigger a bigger relay. This would require soldering

pcornill, 2021-02-14:
Another possible solution with ZMAI-90 but again with Tasmota:
https://www.creationfactory.co/2019/12/zmai-90-din-rail-meterswitch-more.html