xoseperez/espurna #2393: espurna-1.14.1-espurna-base-1MB.bin i'm think i broken is smartplug

Nikovr4, 2020-11-04:
Good day, 
I was only able to flash my TuyaSmartPlug smart socket today, using the tuya-convert project, in espurna, but I don’t know what exactly was asked, as a result, there was no ON / OFF switch in the WEB interface, I wanted to flash tasmota, but via WEB interface I was informed that there was no place for the new firmware. Then I decided to first update the firmware to the smallest one, and then flash the tasmota. As a result, after flashing to espurna-1.14.1-espurna-base-1MB.bin
I see on the router that the device is receiving a wifi network and even pinging commands, but no interfaces are available except telnet.
Tell me how can I get back at least the WEB interface via telnet? Alas, I did not find the UART outputs on the board, I'm afraid that telnet is the only way to recover. Thanks in advance!

Nikovr4, 2020-11-04:
> Good day,
> I was only able to flash my TuyaSmartPlug smart socket today, using the tuya-convert project, in espurna, but I don’t know what exactly was asked, as a result, there was no ON / OFF switch [...]

If it helps, here is the firmware log of my device, thanks!
[http://joxi.ru/Vm6oQElHv4ZJ82](url)

mcspr, 2020-11-04:
Could you also take a closer shot showing the HYS-1-0... text? The top-left of the last photo. UART pins are likely connected to LED & button though, if it is anything like the generic BW-SHP2-like plugs
(btw that you can drag-n-drop images right here)

Can you try flashing the development base .bin after running the `erase.config` in the telnet session? `ota <url>` command then will be able to fetch the new firmware.
Looks like the same things as https://github.com/xoseperez/espurna/issues/2256, I wonder if tuya-convert's espurna.bin is the main culprit

Nikovr4, 2020-11-04:
> Не могли бы вы также сделать более близкий снимок, показывающий HYS-1-0... сообщение? Верхний левый угол последней фотографии. Хотя контакты UART, скорее всего, подключены к светодиоду и кнопке, если это что-то вроде универсальных штекеров типа BW-SHP2
> (кстати, вы можете перетаскивать изображения прямо здесь)
> 
> Можете ли вы попробовать прошить базу разработки ?bin после запуска `erase.config`в сеансе telnet? `ota <url>` команда тогда сможет получить новую прошивку.
> Похоже, то же самое , [что и #2256](https://github.com/xoseperez/espurna/issues/2256), интересно, если туя-новообращенный эспурна.бин-это главный виновник

Thank you for responding to my request, I called PINs, but only 2 of them go to the processor, the rest I could not find, they probably go into obscurity.
Erease.config command does not result in anything, ota command https://github.com/xoseperez/espurna/releases/download/1.14.1/espurna-1.14.1-smartlife-mini-smart-socket.bin
also does not output anything to the console.
any other commands respond:

-ERROR unknown command

mcspr, 2020-11-05:
`erase.config` works just like that, nothing out of ordinary. boards reboots immediately, so there's no output in the terminal
`ota` only works with http in the base firmware, just bring up a local server via node / python / etc.

mcspr, 2020-11-09:
...did it work?

Nikovr4, 2020-11-10:
> ...did it work?

Good day, not worked, i'm install HFS, local web server on http protocol, but command ota http://mypcaddress/firmware.bin not work, in HFS log empty. Other program correctly download file. 
 I think my smart socket is broken permanent...last time is not received IP address and not pinged, i'm lost him:)

closed 2022-05-05 by mcspr