xoseperez/espurna #2396: error compiling my first wemos d1 mini

ramos45, 2020-11-13:
hello,

i try to use ESPurna for the first time with arduino IDE following the guide,adding the libraries etc...

but when i compile my sketch, i have the following  error:

> Arduino:1.8.12 (Windows 7), Tarjeta:"LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

sketch\wifi.cpp: In function 'void wifiSetup()':

sketch\wifi.cpp:768:8: error: 'class JustWifi' has no member named 'begin'

     jw.begin();

        ^

exit status 1
Error compilando para la tarjeta LOLIN(WEMOS) D1 R2 & mini.

Este informe podría contener más información con
"Mostrar salida detallada durante la compilación"
opción habilitada en Archivo -> Preferencias.

i configure the arduino.h uncommenting #define WEMOS_D1_MINI_RELAYSHIELD to see a relay in the web.

 i tried justwifi version 2.0.2, 2.0.1 1.1.3 and 1.1.6

mcspr, 2020-11-13:
Check out https://github.com/xoseperez/espurna/wiki/ArduinoIDE#manual-installation and https://github.com/mcspr/justwifi

ramos45, 2020-11-13:
I follow this guide and use justwifi.

If i try other versions of justwifi, i receive more similar errors

mcspr, 2020-11-13:
That is the point, you can't use any other version but the https://github.com/mcspr/justwifi, it's not available in the library manager

ramos45, 2020-11-13:
I don't understand. I can't ser the problem.

I follow the guide and install .Zip library with this link 

JustWifi | https://github.com/mcspr/justwifi | https://github.com/mcspr/justwifi/archive/2cb9e769.zip
-- | -- | --

In another way, i can see justwifi  the library manager and select the version

Sorry, but my low experience and my bad english difficult it to me

mcspr, 2020-11-13:
> I follow the guide and install .Zip library with this link

This will save the .zip into the %USERPROFILE%\Documents\Arduino\libraries\justwifi-2cb9e7693fc306bf2b15b20132f7904ba11a131a\

> In another way, i can see justwifi the library manager and select the version

This will fetch the original lib and place it into the %USERPROFILE%\Documents\Arduino\libraries\JustWifi\
We don't want this one though.

Now you have 2 libraries, which creates a problem :) Just remove the JustWifi dir, but keep the justwifi-...hash-string..., and try again. Library manager version should not be installed at all, since ArduinoIDE has no way of selecting the specific library version to be used with the 'sketch'.

Note that recommended way to build is with the [PlatformIO](https://github.com/xoseperez/espurna/wiki/PlatformIO) (+ [optionally, VSCode, if you prefer GUI](https://github.com/xoseperez/espurna/wiki/Build-and-update-from-Visual-Studio-Code-using-PlatformIO)). Library management is done automatically there.

ramos45, 2020-11-14:
ok, now it  works!

closed 2020-11-15 by mcspr