xoseperez/espurna #2398: Consider strategic move towards rtl_433 project/library in "RF bridge" section and follow/implement fresh/innovative ideas

osering, 2020-11-14:
Hi!
I'm a regular user of great Espurna on Sonoff switches, but have had limited application/success with such a popular _Sonoff RF Bridge because_ of very narrow number of supported devices (for instance no support for popular/cheap Digoo DG-R8H). And even with new developments and move for using OMG libraries it will not much improve. On the other hand there is great github open source project _rtl_433_ with an extensive (much bigger than OMG's) supported RF devices library.

**Right now there has been important activities in some "merger/cooperation" direction, involving RF Bridge, Portisch (abandoned project), rtl_433 as well as innovate usage of popular RF Bridge.**

Despite Tasmotas' codebase is worse than Espurnas', terms "Tasmota" and "Tasmotized" are still used much more (also in  this recent discussion without anybody mentioning "Espurnized" RF Bridge or Espurna project as such.)

Seeing peoples dislike of present state of RF Bridge potential and planned moves over to rtl_433, coupling Espurna development in this direction with new developments to fill the gap of rtl_433 with ESP devices could be driver for Espurna to gain popularity (be more heard) in comparison with Tasmota.

To have a glimpse of idea,  here are some links with recent developments and discussion with valuable points:
https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/174
https://github.com/halfbakery/RF-Bridge-EFM8BB1/tree/rebaked_sniffer
https://github.com/merbanan/rtl_433/pull/1491
https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/154
https://github.com/merbanan/rtl_433/pull/1542

I see this as really important traction (for all RF hardware users)!

=====

**Is your feature request related to a problem? Please describe.**

Problem: 
1) **Market is still more Tasmotized than Espurnized** :)
2) **Sonoff RF Bridge** (and any other alike products) **support just small number of devices** and people is dropping it in favor of rtl_433 SDR sticks. This deficiency can be used to be the first of competing Tasmota/EspEasy/Esphome projects to **gain more popularity / market share**. 

**Describe the solution you'd like**

**Catch up important developments** quicker than Tasmotas' team and make product better. Move towards using rich rtl_433 library and contemplate idea **separating transport/format to make RF bridge more universal** (preferably supporting all rtl_433 and OMG devices).

mcspr, 2020-12-30:
Note that we kind-of do output the bucket payloads via Portisch fw already, you just need to enable them via the serial command `AAB155`
(I think? Wiki is kind of vague and I don't have rfbridge handy to test - https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xB1)

Commits after https://github.com/xoseperez/espurna/pull/2335 + https://github.com/xoseperez/espurna/pull/2337 should be able to do that:
https://github.com/xoseperez/espurna/blob/47538b9658d36777f81fdc78ce3592f2d8db9e68/code/espurna/rfbridge.cpp#L1115
https://github.com/xoseperez/espurna/blob/47538b9658d36777f81fdc78ce3592f2d8db9e68/code/espurna/rfbridge.cpp#L180-L185

RCSwitch is suboptimal, true. I have thought about porting the bucket approach to the esp8266 itself, but I really wish I'd find some theoretical explanation instead of copying the efm8 source-code before doing anything with that

mcspr, 2021-03-30:
Also note of the https://github.com/NorthernMan54/rtl_433_ESP
Which is currently in use in the OpenMqttGateway, but afaik it is only tested (?) with ESP32